UncleRus/esp-idf-lib #5: Problem with DT Sensor

germacovitch, 2018-09-08:
Hi
I am testing a esp32 board with a DT22 sensor according to the example. However, a get many erros. There is also a correct reading, but most of the times a got a error.
Any idea?
Br
Gilberto

E (161083) DHTxx: Initialization error, problem in phase 'B'
Could not read data from sensor
E (161403) DHTxx: Initialization error, problem in phase 'B'
Could not read data from sensor
E (161723) DHTxx: Initialization error, problem in phase 'B'
Could not read data from sensor
Humidity: 59% Temp: 23C

UncleRus, 2018-09-10:
It looks like some connectivity/wiring problems.
Try to reduce length of  sensor wires.

UncleRus, 2018-09-10:
Did you use a proper pull-up resistor?

Somusundram, 2019-10-18:
> Yes
> The example was using a sample rate that causes error on the reading. I have increased it and the problema was solved.
> BR
> Gilberto

Hi Gilberto,

Just what the value have you increased? I am stuck in here:

E (2387) DHTxx: Initialization error, problem in phase 'C'

afcec, 2019-10-28:
Hi,
I have tested several different libraries with both, Arduino and ESP-IDF, with no success. I am using a PCB that includes a pull-up resistor but with the example provided I was always getting "Initialization error, problem in phase 'B'". I increased the delay to 3,4 and 5 seconds and now the error is "problem in phase 'C'". I am working with ESP32, therefore using pin 17. Any suggestion?

Thanks in advance

jingaworks, 2023-10-29:
I have this same problem

I (11:34:35.021) DHT: Temp  27.2
I (11:34:35.021) DHT: Humid 58.7
I (11:34:40.021) DHT: Temp  27.2
I (11:34:40.021) DHT: Humid 57.3
I (247505) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (247505) wifi:station: d8:f3:bc:67:bd:a1 join, AID=1, bgn, 20
I (11:34:40.061) WIFI_CONNECT: station d8:f3:bc:67:bd:a1 join, AID=1
I (247565) wifi:<ba-add>idx:2 (ifx:1, d8:f3:bc:67:bd:a1), tid:0, ssn:0, winSize:64
I (247565) wifi:<ba-add>idx:3 (ifx:1, d8:f3:bc:67:bd:a1), tid:6, ssn:2, winSize:64
I (11:34:40.113) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
E (11:34:45.016) dht: Initialization error, problem in phase 'B'
E (11:34:45.017) DHT: ret ESP_ERR_TIMEOUT
E (11:34:45.018) DHT: Could not read data from Sensor
I (11:34:45.428) LOCAL_SERVER: New connection opened
E (11:34:50.017) dht: Initialization error, problem in phase 'B'
E (11:34:50.018) DHT: ret ESP_ERR_TIMEOUT
E (11:34:50.019) DHT: Could not read data from Sensor
E (11:34:55.016) dht: Initialization error, problem in phase 'B'
E (11:34:55.017) DHT: ret ESP_ERR_TIMEOUT
E (11:34:55.018) DHT: Could not read data from Sensor
E (11:35:00.016) dht: Initialization error, problem in phase 'B'
E (11:35:00.018) DHT: ret ESP_ERR_TIMEOUT
E (11:35:00.018) DHT: Could not read data from Sensor
E (11:35:05.016) dht: Initialization error, problem in phase 'B'
E (11:35:05.018) DHT: ret ESP_ERR_TIMEOUT
E (11:35:05.018) DHT: Could not read data from Sensor
I (272735) wifi:station: d8:f3:bc:67:bd:a1 leave, AID = 1, bss_flags is 134243, bss:0x3ffbc7f8
I (272735) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (272735) wifi:<ba-del>idx:2, tid:0
I (272745) wifi:<ba-del>idx:3, tid:6
I (11:35:05.284) WIFI_CONNECT: station d8:f3:bc:67:bd:a1 leave, AID=1
E (11:35:10.016) dht: Initialization error, problem in phase 'B'
E (11:35:10.017) DHT: ret ESP_ERR_TIMEOUT
E (11:35:10.018) DHT: Could not read data from Sensor
I (11:35:15.021) DHT: Temp  26.9
I (11:35:15.022) DHT: Humid 52.7
E (11:35:16.416) ws_keep_alive: Client (fd=57) not alive!
E (11:35:16.417) LOCAL_SERVER: Client not alive, closing fd 57
I (11:35:20.021) DHT: Temp  26.8
I (11:35:20.022) DHT: Humid 52.8
I (11:35:25.020) DHT: Temp  26.8
I (11:35:25.021) DHT: Humid 53.3

This happens only when when I'm connected to WIFI AP mode.
If I'm not connected to AP, everything works normal.

I use pin 17.

UncleRus, 2023-10-30:
Read the [FAQ](https://github.com/UncleRus/esp-idf-lib/blob/master/FAQ.md#how-can-i-include-ets_sysh-in-my-code-without-complex-macros-and-remain-compatible-with-different-targets).