UncleRus/esp-idf-lib #6: Usind multiple sensors

tilman, 2019-03-08:
Is it possible to use multiple sensors with the same port? I tried to use a bmp280 and a tsl2561 at the same i2c port but always got errors during init. Using them standalone worked fine. Would be great if you could provide an example of using 2 sensors.

tilman, 2019-03-08:
Okay, got it. It is not possible to init 2 sensors on the same thread correct? You have to use multiple threads/tasks.

colesnicov, 2019-08-07:
Hi @tilman, have you made any progress with this problem? Thank you for answer

UncleRus, 2019-08-08:
Hi @colesnicov!

I added an example of working with several I2C devices on same bus (BMP280 and PCF8574), see examples/simplest_barometer.
Also you can see ads1115 example.

colesnicov, 2019-08-08:
Я заметил, что в подпрограммах i2cdev есть семафор, чтобы предотвратить что? Несанкционированный доступ к линии I2C при связи с другим устройством? 

Я задавался вопросом об этом, и что мне, что, возможно, нет необходимости.? Когда я прочитал [это](https://www.reddit.com/r/esp32/comments/9t14xf/help_troubleshooting_occasional_lockups_for_a/e8sz7ou/),  и [здесь](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/api-reference/peripherals/spi_master.html#thread-safety), а также после некоторого общения [здесь](https://github.com/espressif/esp-iot-solution/issues/35), я не уверен, что это необходимо. Что вы думаете? 

Это правильно или необходимо? Поскольку i2cdev использует собственный (esp-idf) драйвер I2C, который не является потокобезопасным, но безопасным для задач. Я еще не смотрел, как то, что представляет собой функцию port_ENTER_CRITICAL, но оно должно быть достаточно. Это также зависит от реализации...

Я хотел бы понять это прежде, чем обернуть это в "коробку" ..

UncleRus, 2019-08-09:
Да, в i2cdev используются два типа мьютексов: портовые и транзакционные.

Портовые мьютексы (их всего два, по количеству портов I2C) необходимы, чтобы избежать проблем в ситуации, упомянутой в документации:

> The I2C APIs are not thread-safe, if you want to use one I2C port in different tasks, you need to take care of the multi-thread issue. 

Мьютексы второго типа создаются по одному на устройство и являются необходимыми в случае, если одно устройство используется в нескольких задачах:

```C
static i2c_dev_t some_dev;

void task1 (void *arg)
{
     // some_dev used here
}

void task2 (void *arg)
{
     // and here some_dev used too
}
```

UncleRus, 2019-08-10:
I leave this thread open.

aledoors, 2020-08-01:
Hello friends,

Can i connect 2 o 3 BMP280 sensors on the same port of I2C?

UncleRus, 2020-08-01:
You either need to use different addresses on these sensors and connect them to the common SDA and SCL GPIOs, or you have to use separate GPIOs for these sensors, for example:

```C
// First sensor
#define SENSOR0_SDA_GPIO 16
#define SENSOR0_SCL_GPIO 17
// Second sensor
#define SENSOR1_SDA_GPIO 21
#define SENSOR1_SCL_GPIO 22

...

bmp280_t dev0, dev1;
memset(&dev0, 0, sizeof(bmp280_t));
memset(&dev1, 0, sizeof(bmp280_t));

// Init first sensor
ESP_ERROR_CHECK(bmp280_init_desc(&dev0, BMP280_I2C_ADDRESS_0, 0, SENSOR0_SDA_GPIO, SENSOR0_SCL_GPIO));
ESP_ERROR_CHECK(bmp280_init(&dev0, &params0));
// Init second sensor
ESP_ERROR_CHECK(bmp280_init_desc(&dev1, BMP280_I2C_ADDRESS_0, 0, SENSOR1_SDA_GPIO, SENSOR1_SCL_GPIO));
ESP_ERROR_CHECK(bmp280_init(&dev1, &params1));
....
// Use sensors
if (bmp280_read_float(&dev0, &temperature, &pressure, &humidity) == ESP_OK)
        printf("Sensor0 pressure: %.2f Pa, temperature: %.2f C", pressure, temperature);
if (bmp280_read_float(&dev1, &temperature, &pressure, &humidity) == ESP_OK)
        printf("Sensor1 pressure: %.2f Pa, temperature: %.2f C", pressure, temperature);

```

aledoors, 2020-08-01:
Uncle Rus, thanks for your help! I'm doing a project, have to connect 3 BMP280 sensors at most in one Esp32 (slave), but, maybe 2 or 1 sensors too and send the information about temperature and pressure to another Esp32 (master). I was thinking an I2C connection between all, is it possible? I don't want use wifi connection.

closed 2020-09-04 by UncleRus

bastianhjaeger, 2021-01-25:
Hi, 
I saw this thread issue and maybe my question fits here best. Due to the conversation so far I am not sure on the best way to implement 
- with 1 I²C port and
- 2 or more devices connected to it. 

If my two devices work in the **same or maybe different threads**, the should use the same `i2c_dev_t` instance for the mutex, right?

In @UncleRus example in "examples/simplest_barometer" is see that 
   `pcf8574_init_desc()`
and 
   `bmp280_init_desc()`
do each have their own instance of` i2c_dev_t`

Important question: Should two physical seinsor/actuator I²C devices on the same I²C port use the same instance of `i2c_dev_t` for the mutex or not?

Thanks

bastianhjaeger, 2021-01-26:
I answered the question myself (partly). 

There is the macro `SEMAPHORE_TAKE(dev->port);` used in `i2c_dev_read(...)` and `i2c_dev_write(...)` that makes sure the port is only used by one device. 
But just to be sure: Is there any reason to use two instances of `i2c_dev_t` or could you as well use one one instance for two physical sensors?

juand2207, 2023-01-10:
Hello friends, I have a project that I must carry out connecting two dht11 temperature sensors to the same ESP32. Could you give me any suggestions to start?