UncleRus/esp-idf-lib #7: ds1307 example xQueueGenericReceive - assertion

marcogmaia, 2019-03-22:
No matter what I do, I can't get the example of the ds1307 to run. always gives me the following:

[...]
I (100) cpu_start: Starting scheduler on PRO CPU.
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (10) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (10) cpu_start: Starting scheduler on APP CPU.
D (150) heap_init: New heap initialised at 0x3ffe0440
D (150) heap_init: New heap initialised at 0x3ffe4350
V (160) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (160) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (160) intr_alloc: Connected src 16 to int 12 (cpu 0)
V (180) I2C_DEV: [0x68 at 0] creating mutex
V (180) I2C_DEV: [0x68 at 0] taking mutex
/home/marco/dev/iot/esp32/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40085500 on core 1

I am running the example unmodified.
[ds1307](https://github.com/UncleRus/esp-idf-lib/tree/master/examples/ds1307/main/main.c)

UncleRus, 2019-04-04:
I'm sorry, but I can't reproduce this error with the latest esp-idf.

What version of esp-idf do you use?

marcogmaia, 2019-04-05:
This happened when I updated my esp-idf, before the update it was working; I'll update again and try.

marcogmaia, 2019-04-05:
Had no success, I think it is possible to be a wrong config in my freeRTOS menuconfig

UncleRus, 2019-04-05:
Please feel free to reopen

ansemjo, 2020-08-25:
I think I am hitting this same error just now. The relevant code looks like this:

```c
// display the current runtime in MM:SS on display
void rtctime(void *arg) {

  vfd_handle_t *vfd = arg;
  
  i2c_dev_t rtc;
  memset(&rtc, 0, sizeof(i2c_dev_t));

  ESP_ERROR_CHECK(ds1307_init_desc(&rtc, I2C_NUM_0, 21, 22));
  ESP_LOGI("rtctime", "initialized ...");

  ESP_ERROR_CHECK(ds1307_start(&rtc, true));
  ESP_LOGI("rtctime", "rtc started ...");

  // time_t now;
  struct tm time;
  char timebuf[32];

  while (true) {

    // ERROR: hitting "(xQueueGenericReceive)- assert failed!" here
    // when trying to take the mutex
    if (ds1307_get_time(&rtc, &time) != ESP_OK) {
      printf("couldn't get time from rtc");
    } else {
      strftime(timebuf, sizeof(timebuf), "%M:%S", &time);
      printf("%s", timebuf);
      vfd_text(vfd, timebuf);
    };

    vTaskDelay(100 / portTICK_RATE_MS);
  }
}
```

And the debug log shows this:

```
V (1306) I2C_DEV: [0x68 at 0] creating mutex
I (1316) rtctime: initialized ...
V (1316) I2C_DEV: [0x68 at 0] taking mutex
/home/ansemjo/.platformio/packages/framework-espidf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40087dc4  PS      : 0x00060c30  A0      : 0x800eae3f  A1      : 0x3ffb7db0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000064  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000005  A8      : 0x80087dc1  A9      : 0x3ffb7d20  
A10     : 0x0000007c  A11     : 0x0000007c  A12     : 0x00000010  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xffffffeb  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
```

This is a custom board based on an ESP32-WROOM-32E and I am using the ESP-IDF wrapped in a [PlatformIO project](https://github.com/ansemjo/chronovfd/tree/11d1aa90a20e109af0aa46bbe181a2175929d32b/firmware). Note that the [i2cscan](https://github.com/ansemjo/chronovfd/blob/11d1aa90a20e109af0aa46bbe181a2175929d32b/firmware/src/main.c#L34) function copied from the ESP-IDF API examples works fine.

Tristaan, 2021-07-13:
A similar problem with bmp280.
The error is as follows:
```
/.platformio/packages/framework-espidf/components/freertos/queue.c:1447 (xQueueGenericReceive)- assert failed!

abort() was called at PC 0x400883dd on core 0
```

platformio is used and the versions are:
```
 - framework-espidf 3.40201.210513 (4.2.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.7.1
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
 ```
I've done some backtracing and found that this happens inside `i2c_dev_read()`. I am suspicious of the SEMAPHORE_TAKE(),  macro, because right after i2c_dev_read() the xQueueGenericReceive() is called, but i couldn't find any references to xQueueGenericReceive().

UncleRus, 2021-07-13:
Yep, this is known bug with ESP-IDF versions below 4.0.
What can you do:
- Update ESP-IDF (recommended)
- Disable mutexes in `i2cdev` as described here: https://github.com/UncleRus/esp-idf-lib/blob/master/FAQ.md#can-i-use-i2c-device-drivers-from-interrupts

ansemjo, 2021-07-13:
> Update ESP-IDF (recommended, v3.x is waaaaay too old)

But both mine (`framework-espidf 3.40001.200521 (4.0.1)`) and @Tristaan's version (`framework-espidf 3.40201.210513 (4.2.1)`) is > 4.0 (note the version in parentheses .. the `3.40...` is just PlatformIO's versioning "alias" for it)? Or am I misunderstanding something?

UncleRus, 2021-07-13:
Hmm. Let's see: the assert on line 1447 of queue.c looks like this:
```C
/ * Check the queue pointer is not NULL. * /
configASSERT ((pxQueue));
```
Its argument (pxQueue) is a mutex from I2C device descriptor. Basically this assertion just checks if i2c_dev.mutex != NULL. And it can't be NULL cause it created successfully in `ds1307_init_desc()` function.

And the strangest thing is that I can't reproduce this bug at all.
Literally just assembled a simple installation from esp32 and ds1307, uploaded an example and it works great.
```
...
I (319) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
2018-04-11 00:52:10
2018-04-11 00:52:11
2018-04-11 00:52:12
2018-04-11 00:52:12
2018-04-11 00:52:13
...
```

Tristaan, 2021-07-14:
So, after disabling the mutexes with CONFIG_I2CDEV_NOLOCK there is another error, this happens on `bmp280_init()`:
```
i2c: i2c_param_config(645): i2c clock choice is invalid, please check flag and frequency
```
After browsing the code, I changed the I2C_FREQ_HZ variable to no avail. Then tried to add defines for the HELPER_TARGET_IS_ESP32, also nothing. So something is not correct, but this is a whole other bug. After comparing the code from ds1307 with bmp280 it is the same and should work. Reading the sources from esp-idf, [L640](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/driver/i2c.c#L640) is the culprit. Can this be caused by the connected bmp280? But the clk gets checked before any real device writing and reading is done. It seems the only way to go is JTAG.