UncleRus/esp-idf-lib #10: (question) on SGP30 and BME680

skitsanos, 2019-04-15:
Hi, not an issue but rather a question, do you have any plans to add support for SGP30 and BME680 sensors?

UncleRus, 2019-04-16:
Hi!
There is already a driver for SGP30 developed by it's manufacturer (BSD-3 license) 
https://github.com/Sensirion/embedded-sgp
You can easily create component for esp-idf and all your need is `sensirion_hw_i2c_implementation.c`, which might looks like
```C
#include <i2cdev.h>
#include <esp_log.h>

static const char *TAG = "SGP";

static i2c_dev_t dev;

/**
 * Initialize all hard- and software components that are needed for the I2C
 * communication.
 * You need to define SGP_PORT, SGP_SCL_GPIO and SGP_SDA_GPIO somewhere
 * Do not call this function more than once!
 */
void sensirion_i2c_init()
{
    dev.port = SGP_PORT;
    dev.address = 0x58;
    dev.cfg.sda_io_num = SGP_SDA_GPIO;
    dev.cfg.scl_io_num = SGP_SCL_GPIO;
    dev.cfg.master.clk_speed = 400000; // 400kHz
    ESP_ERROR_CHECK(i2c_dev_create_mutex(&dev));
}

s8 sensirion_i2c_read(u8 address, u8* data, u16 count)
{
    if (i2c_dev_take_mutex(&dev) != ESP_OK) return -1;
    if (i2c_dev_read(&dev, NULL, 0, data, count) != ESP_OK)
    {
        ESP_LOGE(TAG, "Could not read from device");
        return -1;
    }
    if (i2c_dev_give_mutex(&dev) != ESP_OK) return -1;
    return 0;
}

s8 sensirion_i2c_write(u8 address, const u8* data, u16 count)
{
    if (i2c_dev_take_mutex(&dev) != ESP_OK) return -1;
    if (i2c_dev_write(&dev, NULL, 0, data, count) != ESP_OK)
    {
        ESP_LOGE(TAG, "Could not write to device");
        return -1;
    }
    if (i2c_dev_give_mutex(&dev) != ESP_OK) return -1;
    return 0;
}

```

UncleRus, 2019-04-16:
BME680 on my to-do list, but unfortunately I can't say for sure when I can do it.
Any help would be appreciated.

UncleRus, 2019-04-27:
Ok, there is a driver for BME680, https://github.com/gschorcht/bme680-esp-idf
I leave this thread open in case someone will also look for drivers for these devices.

UncleRus, 2019-08-19:
I adapted this BME680 driver for use with i2cdev and created pull request: https://github.com/UncleRus/esp-idf-lib/pull/29

Would you test this driver? I don't have BME680 right now.

closed 2020-09-04 by UncleRus