UncleRus/esp-idf-lib #12: mcp23017 library problam

kshyshkin, 2019-05-21:
Hello. Have some problem with `mcp23017_port_write()` function. When i call it i see in monitor
`E (751) i2c: /home/kshyshki/esp/esp-idf/components/driver/i2c.c:335 (i2c_driver_delete):i2c driver install error
E (771) i2c: /home/kshyshki/esp/esp-idf/components/driver/i2c.c:870 (i2c_set_pin):this i2c pin does not support internal pull-up
`
Try difference pins for i2c (21,22; 16,17 (as in your example)), problem the same...

`mcp23017_port_write()` return `ESP_ERR_INVALID_ARG,`

Thank you for advance for your reply.

kshyshkin, 2019-05-22:
Problem was because in `https://github.com/UncleRus/esp-idf-lib/blob/master/examples/mcp23017/main/main.c#L20` after initialization was garbage, in `dev.cfg.sda_pullup_en` and `dev.cfg.scl_pullup_en` was `1`. After i add after initialization `memset(&dev, 0, sizeof(i2c_dev_t));`, all return in normal condition. But i dont understand why this error happens in pull-up enabled.

kshyshkin, 2019-05-22:
One more question:
Is it possible use your library with 2 or more mcp23017 with different addresses at the same time? (0x20, 0x21, 0x22).
If yes, could you provide simple example?
I try but get crash by freeRTOS near mutex of i2c port.

UncleRus, 2019-07-11:
Thank you! I've updated all examples.

UncleRus, 2019-07-12:
And I fixed another bug in this library: 3e6a3fd

kshyshkin, 2019-09-20:
Hello. In first message i write that have some error like ```E (751) i2c: /home/kshyshki/esp/esp-idf/components/driver/i2c.c:335 (i2c_driver_delete):i2c driver install error``` this error still present. Please you answer is it normal ? or no?

closed 2020-05-02 by UncleRus