UncleRus/esp-idf-lib #13: compile error examples/ds3231 and examples/ds1307

nopnop2002, 2019-05-27:
I tried to compile these.
esp-idf-lib/examples/ds3231 and esp-idf-lib/examples/ds1307
But compile error occure.

	/home/nop/esp-idf-lib/components/hd44780/hd44780.c: In function 'write_nibble':
	/home/nop/esp-idf-lib/components/hd44780/hd44780.c:47:29: error: implicit declaration of function 'ets_delay_us' [-Werror=implicit-function-declaration]
	 #define toggle_delay() do { ets_delay_us(DELAY_TOGGLE); } while (0)
	                             ^
	/home/nop/esp-idf-lib/components/hd44780/hd44780.c:65:9: note: in expansion of macro 'toggle_delay'
	         toggle_delay();
	         ^
	/home/nop/esp-idf-lib/components/hd44780/hd44780.c: At top level:
	cc1: warning: unrecognized command line option '-Wno-frame-address'
	cc1: some warnings being treated as errors
	/home/nop/esp-idf/make/component_wrapper.mk:289: recipe for target 'hd44780.o' failed
	make[1]: *** [hd44780.o] Error 1
	/home/nop/esp-idf/make/project.mk:582: recipe for target 'component-hd44780-build' failed
	make: *** [component-hd44780-build] Error 2

So i added this line.
It's resolve.

sp-idf-lib/components/hd44780/hd44780.c

	/**
	 * ESP-IDF driver for HD44780 compartible LCD text displays
	 *
	 * Ported from esp-open-rtos
	 * Copyright (C) 2016, 2018 Ruslan V. Uss <unclerus@gmail.com>
	 * BSD Licensed as described in the file LICENSE
	 */
	#include "hd44780.h"
	#include <string.h>
	#include <esp_system.h>
	#include <esp32/rom/ets_sys.h> // add by nopnop2002

esp-idf-lib/components/max7219/max7219.h

	/**
	 * @file max7219.h
	 *
	 * ESP-IDF driver for MAX7219/MAX7221
	 * Serially Interfaced, 8-Digit LED Display Drivers
	 *
	 * Ported from esp-open-rtos
	 * Copyright (C) 2017, 2018 Ruslan V. Uss <unclerus@gmail.com>
	 * BSD Licensed as described in the file LICENSE
	 */
	#ifndef __MAX7219_H__
	#define __MAX7219_H__

	#include <stdint.h>
	#include <stdbool.h>
	#include <driver/spi_master.h>
	#include <driver/gpio.h> // add by nopnop2002

jcabad100, 2019-05-29:
i have exact the same error (rror: implicit declaration of function 'ets_delay_us' [-Werror=implicit-function-declaration]
 #define toggle_delay())  trying other examples/sensors like the DHT ones., more than one week reported in separated issue..so far nobody has helped me to solve it....

UncleRus, 2019-06-01:
@nopnop2002 Could you make a pull request?

nopnop2002, 2019-06-03:
@UncleRus
I made pull request.
When you commit it,please close this issues.

closed 2019-06-03 by UncleRus