UncleRus/esp-idf-lib #23: Im CMakeLists noob

kuba1342, 2019-08-02:
Hello, when I compile the project I get:
- undefined reference to `ds1302_init'
- undefined reference to `ds1302_set_write_protect'
- undefined reference to `ds1302_get_time'

Do you have any ideas what I could have missed?

UncleRus, 2019-08-02:
Hi!
Did you add path to the components directory to EXTRA_COMPONENTS_DIR in your projects CMakeLists.txt?
```CMakeLists
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS /path/to/esp-idf-lib/components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(example-ds1302)
```

kuba1342, 2019-08-02:
Okay, I see like there is problem with my IDF_PATH

`export IDF_PATH="$HOME/Pulpit/Greengrass/GitV/amazon-freertos/vendors/espressif/esp-idf"
`

I checked twice, path is correct but it still throws

`CMake Error at CMakeLists.txt:37 (include):
  include could not find load file:

    /tools/cmake/project.cmake`

kuba1342, 2019-08-02:
Okay, I changed path but didnt restart CLion. Now its working, thanks

closed 2019-08-02 by kuba1342