UncleRus/esp-idf-lib #42: mcp23017 example problem

jponko, 2019-09-18:
Tried to compile the mcp23017 example program for a esp32 lyrat board but it doesn't appear to be compatible with esp-idf v3.2.2. I looked at dht.c and it's looking for a define for CONFIG_IDF_TARGET_ESP32 but I thought I read that this isn't for the esp8266 anyway. There's no mention of any special config steps needed in the README.md files. There are several other files that are also looking for this define. Did I miss something?

CC build/esp32/hwcrypto/aes.o
AS build/esp32/dport_panic_highint_hdl.o
AR build/esp32/libesp32.a
CC build/dht/dht.o
/home/jponko/esp/esp-idf-lib/components/dht/dht.c: In function 'dht_read_data':
/home/jponko/esp/esp-idf-lib/components/dht/dht.c:177:5: error: 'PORT_ENTER_CRITICAL' undeclared (first use in this function)
     PORT_ENTER_CRITICAL;
     ^
/home/jponko/esp/esp-idf-lib/components/dht/dht.c:177:5: note: each undeclared identifier is reported only once for each function it appears in
/home/jponko/esp/esp-idf-lib/components/dht/dht.c:179:5: error: 'PORT_EXIT_CRITICAL' undeclared (first use in this function)
     PORT_EXIT_CRITICAL;
     ^
/home/jponko/esp/esp-idf-v3.2.2/make/component_wrapper.mk:285: recipe for target 'dht.o' failed
make[1]: *** [dht.o] Error 1
/home/jponko/esp/esp-idf/make/project.mk:530: recipe for target 'component-dht-build' failed
make: *** [component-dht-build] Error 2

trombik, 2019-09-18:
in `master` barnch of `esp-idf`, `CONFIG_IDF_TARGET_ESP32` is defined in `sdkconfig`. probably it was introduced after `v3.2.2`. will see if the library can supports older releases. it would take a while as i have to patch `esp-idf` first to create a working build environment (my OS is supported only in master).

trombik, 2019-09-18:
i agree that `README.md` should mention this.

trombik, 2019-09-18:
not tested, but manually defining `CONFIG_IDF_TARGET_ESP32` (in `Makefile` or environment variable) should work without ill effect.

jponko, 2019-09-18:
I was able to compile it by maunally adding the defines to the affected
files but after flashing it crashed with a guru meditation error:

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6188
load:0x40078000,len:10180
load:0x40080400,len:6660
entry 0x40080764
I (29) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloaderr
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0a37c
( 41852) map
I (108) esp_image: segment 1: paddr=0x0001a3a4 vaddr=0x3ffb0000
size=0x01ecc (  7884) load
I (111) esp_image: segment 2: paddr=0x0001c278 vaddr=0x40080000
size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at
/home/jponko/esp/esp-idf-v3.2.2/components/freertos/xtensa_vectors.S:1779

I (115) esp_image: segment 3: paddr=0x0001c680 vaddr=0x40080400
size=0x03990 ( 14736) load
I (130) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018
size=0x14ebc ( 85692) map
0x400d0018: _flash_cache_start at ??:?

I (163) esp_image: segment 5: paddr=0x00034edc vaddr=0x40083d90
size=0x06cc4 ( 27844) load
0x40083d90: spi_flash_munmap at
/home/jponko/esp/esp-idf-v3.2.2/components/spi_flash/flash_mmap.c:319

I (181) boot: Loaded app from partition at offset 0x10000
I (181) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (185) cpu_start: Starting app cpu, entry point is 0x40080f40
0x40080f40: call_start_cpu1 at
/home/jponko/esp/esp-idf-v3.2.2/components/esp32/cpu_start.c:246

I (177) cpu_start: App cpu up.
I (196) heap_init: Initializing. RAM available for dynamic allocation:
I (202) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (208) heap_init: At 3FFB2F60 len 0002D0A0 (180 KiB): DRAM
I (215) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (221) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (227) heap_init: At 4008AA54 len 000155AC (85 KiB): IRAM
I (234) cpu_start: Pro cpu start user code
I (252) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (0) i2c: /home/jponko/esp/esp-idf-v3.2.2/components/driver/i2c.c:335
(i2c_driver_delete):i2c driver install error
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception
was unhandled.
Core 1 register dump:
PC      : 0x400e460f  PS      : 0x00060933  A0      : 0x800e3501  A1      :
0x3ffb7a00
0x400e460f: i2c_param_config at
/home/jponko/esp/esp-idf-v3.2.2/components/driver/i2c.c:661

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb1264  A5      :
0x3ffb7a48
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ff53000  A9      :
0x04c4b400
A10     : 0x0155408b  A11     : 0x00000003  A12     : 0x00000011  A13     :
0x00000a00
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000000f  EXCCAUSE:
0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  :
0xfffffff9

Backtrace: 0x400e460f:0x3ffb7a00 0x400e34fe:0x3ffb7a30
0x400e368e:0x3ffb7a70 0x400d257e:0x3ffb7ab0 0x400d2652:0x3ffb7ae0
0x400d2435:0x3ffb7b00 0x40086061:0x3ffb7b50
0x400e460f: i2c_param_config at
/home/jponko/esp/esp-idf-v3.2.2/components/driver/i2c.c:661

0x400e34fe: i2c_setup_port at
/home/jponko/esp/esp-idf-lib/components/i2cdev/i2cdev.c:94

0x400e368e: i2c_dev_read at
/home/jponko/esp/esp-idf-lib/components/i2cdev/i2cdev.c:175

0x400d257e: i2c_dev_read_reg at
/home/jponko/esp/esp-idf-lib/components/mcp23x17/mcp23x17.c:401
 (inlined by) write_reg_bit_16 at
/home/jponko/esp/esp-idf-lib/components/mcp23x17/mcp23x17.c:83

0x400d2652: mcp23x17_set_mode at
/home/jponko/esp/esp-idf-lib/components/mcp23x17/mcp23x17.c:401

0x400d2435: test at
/home/jponko/esp/esp-idf-lib/examples/mcp23017/main/main.c:29

0x40086061: vPortTaskWrapper at
/home/jponko/esp/esp-idf-v3.2.2/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

trombik, 2019-09-18:
did you`make clean` or `rm -rf ./build` before compiling?

jponko, 2019-09-18:
A make clean doesn't make any difference.

trombik, 2019-09-18:
what if you define it in `examples/mcp23017/Makefile`?

jponko, 2019-09-18:
i deleted the esp-idf-lib, cloned it again and then defined

Now the CPU isn't crashing but it looks like the mcp23017 can't be read
over the i2c bus:

I (0) cpu_start: App cpu up.
I (198) heap_init: Initializing. RAM available for dynamic allocation:
I (205) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (211) heap_init: At 3FFB2F60 len 0002D0A0 (180 KiB): DRAM
I (217) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (223) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (230) heap_init: At 4008AA54 len 000155AC (85 KiB): IRAM
I (236) cpu_start: Pro cpu start user code
I (254) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (0) i2c: /home/jponko/esp/esp-idf-v3.2.2/components/driver/i2c.c:335
(i2c_driver_delete):i2c driver install error
E (10) I2C_DEV: Could not read from device [0x20 at 0]: 263
E (10) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (20) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (20) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (1315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (1815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (2315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (2815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (3315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (3815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (4315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (4815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (5315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (5815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (6315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (6815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (7315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (7815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (8315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (8815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (9315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (9815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (10315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (10815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (11315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (11815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (12315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (12815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (13315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (13815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (14315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (14815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (15315) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (15815) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (16315) I2C_DEV: Could not read from device [0x20 at 0]: -1

trombik, 2019-09-18:
errno 263 is `ESP_ERR_TIMEOUT`. this is a different issue, i guess. this happens in many cases: the device is powered off, wrong wiring, missing pull-up register, etc.

trombik, 2019-09-18:
created a branch to workaround the issue. please clone it by:

```
git clone --branch matrix_idf_version https://github.com/trombik/esp-idf-lib
```
then, add the following line to `examples/mcp23017/Makefile` (i cannot test on my local machine at the moment).

```
PROJECT_CONFIG_IDF_TARGET_ESP32 = y
```
if it does not work, try:

```
CFLAGS += -DPROJECT_CONFIG_IDF_TARGET_ESP32
```

then, run `make` as usual.

jponko, 2019-09-18:
These won't work. However, if I add CFLAGES += CONFIG_IDF_TARGET_ESP32" to
the makefile that works as I demonstrated before. Even so, there's a
unrelated  issue with ets_sys.h not being found in some of the other
drivers. So I put the full path to ets_sys.h in ds1302.c and hd44780.c to
fix the problem:

//#include <esp32/rom/ets_sys.h>// add by nopnop2002
#include "/home/jponko/esp/esp-idf/esp32/rom/ets_sys.h"

However, this doesn't change the fact that the mcp23017 driver still
doesn't work even if I can get it to compile and flash.

Do you have a mcp23017 module to try this driver? I found another driver
that worked with the mcp23017 but unfortunately it's not thread safe. I
don't think it has anything to do with the i2c driver as that worked with
the ADS1115 example. The reason I became interested in the mcp23017 driver
was because of the thread safe i2c driver and I didn't want to reinvent the
wheel. I seems most i2c device example programs I've seen aren't thread
safe because they are single threaded.

trombik, 2019-09-19:
`ets_sys.h` header issues have been fixed in the branch i mentioned, and merged into `master` by #43 

correct, `i2cdev` works in other drivers.

please try the latest master. the commit [passed the test](https://travis-ci.org/UncleRus/esp-idf-lib/jobs/586565063) using esp-idf  v3.2.2. the build process uses `CFLAGS` method, but PROJECT_CONFIG_IDF_TARGET_ESP32 is supposed to be the correct workaround.

no, I am currently on vacation and I did not bring the chip with me. have you looked at the signal on the bus with oscilloscope?

jponko, 2019-09-19:
I tried the latest master last night and using
PROJECT_CONFIG_IDF_TARGET_ESP32 which works.
I don't think looking at the signal with an oscilloscope is going to tell
me much. I may purchase a Bus Pirate to capture the protocol to see what's
being sent.

In the meantime, I'll play around with the mcp23017 example.

trombik, 2019-09-23:
the latest master has a better fix. you should br able to build without the workaround mentioned in this issue.

closed 2019-09-23 by trombik