UncleRus/esp-idf-lib #46: I2c problems

winny2809, 2019-10-15:
Is it possible to have some examples regarding the i2c device and if possible also of the device pcf8575?

I can't understand many things about it, and I would have some examples, maybe even an initial configuration :-(

UncleRus, 2019-10-15:
See [example](https://github.com/UncleRus/esp-idf-lib/tree/master/examples/pcf8575)

winny2809, 2019-10-15:
Thank you very much for the timeliness with which you helped me, very kind.
I integrated his library into my esp-idf system in Platformio, it is, unfortunately it generates me a lot of errors and I can not find the cause.
Could you tell me what I'm wrong?

[image]

trombik, 2019-10-16:
i cannot tell where the issue is, but looks to me that your build system is not correctly configured. more information would certainly help.

* full build log in text (not screen shot)
* `platformio.ini`

by the way, `platformio` does not support `esp-idf` very well, e.g. `sdkconfig` handling and other build-related issues. you need to understand and translate various build flags. the tool-chains they ships are often lagged behind.

winny2809, 2019-10-16:
Below, I attached the results of the compilation.
Of course, as you say, platformer has problems especially with regard to "sdkconfig". Usually, however, I solve by inserting the various variables within the C file, such as in this case I put in the file i2cdev.c  this definition: #define CONFIG_I2CDEV_TIMEOUT 1000
Unfortunately however, I do not understand the cause of all these mistakes, it will definitely be a nonsense to solve, but you need someone with more experience than me.

> Executing task: C:\Users\Winny\.platformio\penv\Scripts\platformio.exe run <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30300.190916 (3.3.0), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <pcf8575>
|   |-- <i2cdev>
Compiling .pio\build\esp32doit-devkit-v1\src\main.o
In file included from lib/pcf8575/pcf8575.h:31:0,
                 from src\main.c:8:
lib/i2cdev/i2cdev.h:285:43: error: 'dev' undeclared here (not in a function)
         esp_err_t __ = i2c_dev_take_mutex(dev); \
                                           ^
lib/i2cdev/i2cdev.h:287:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:289:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:289:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:297:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:299:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:299:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:307:9: error: expected identifier or '(' before 'if'
         if (___ != ESP_OK) { \
         ^
lib/i2cdev/i2cdev.h:315:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
In file included from C:/Users/Winny/.platformio/packages/framework-espidf/components/esp32/include/esp_timer.h:44:0,
                 from C:/Users/Winny/.platformio/packages/framework-espidf/components/freertos/include/freertos/portmacro.h:82,
                 from C:/Users/Winny/.platformio/packages/framework-espidf/components/freertos/include/freertos/portable.h:94,
                 from C:/Users/Winny/.platformio/packages/framework-espidf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from src\main.c:4:
src\main.c: In function 'test':
src\main.c:50:21: error: implicit declaration of function 'pcf8575_init_desc' [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(pcf8575_init_desc(&pcf8575, 0, I2C_ADDR, SDA_GPIO, SCL_GPIO));
                     ^
C:/Users/Winny/.platformio/packages/framework-espidf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
cc1.exe: some warnings being treated as errors
Compiling .pio\build\esp32doit-devkit-v1\lib6ca\i2cdev\i2cdev.o
Compiling .pio\build\esp32doit-devkit-v1\lib588\pcf8575\pcf8575source.o
*** [.pio\build\esp32doit-devkit-v1\src\main.o] Error 1
In file included from lib\i2cdev\i2cdev.c:11:0:
lib\i2cdev\i2cdev.h:285:43: error: 'dev' undeclared here (not in a function)
         esp_err_t __ = i2c_dev_take_mutex(dev); \
[... 156 lines cut ...]
     } while (0)
     ^
lib/i2cdev/i2cdev.h:289:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:297:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:299:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:299:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:307:9: error: expected identifier or '(' before 'if'
         if (___ != ESP_OK) { \
         ^
lib/i2cdev/i2cdev.h:315:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
*** [.pio\build\esp32doit-devkit-v1\lib588\pcf8575\pcf8575source.o] Error 1
==================================================== [FAILED] Took 26.14 seconds ====================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

winny2809, 2019-10-16:
The main errors I find in these lines of code:

#define I2C_DEV_TAKE_MUTEX(dev) do { \

        esp_err_t __ = i2c_dev_take_mutex(dev); \

        if (__ != ESP_OK) return __;\

    } while (0)

#define I2C_DEV_GIVE_MUTEX(dev) do { \

        esp_err_t __ = i2c_dev_give_mutex(dev); \

        if (__ != ESP_OK) return __;\

    } while (0)

#define I2C_DEV_CHECK(dev,X) do { \

        esp_err_t ___ = X; \

        if (___ != ESP_OK) { \

            I2C_DEV_GIVE_MUTEX(dev); \

            return ___; \

        } \

    } while (0)

and also in the definitions of  SEMAPHORE...
It seems to me that the problems come from SEMAPHORE or at least from The Freertos.
In my file c_cpp_properties.json i have this code:

"intelliSenseMode": "clang-x64",
            "cStandard": "c99",
            "cppStandard": "c++11",
            "compilerPath": "\"C:/Users/Winny/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe\" -mlongcalls"

I hope this will help you understand my problem

trombik, 2019-10-16:
what would happen if you add `#include <esp_err.h>` to `lib/i2cdev/i2cdev.h`? something like:

```diff
diff --git a/components/i2cdev/i2cdev.h b/components/i2cdev/i2cdev.h
index ba68248..63e75d5 100644
--- a/components/i2cdev/i2cdev.h
+++ b/components/i2cdev/i2cdev.h
@@ -15,6 +15,7 @@
 #include <driver/i2c.h>
 #include <freertos/FreeRTOS.h>
 #include <freertos/semphr.h>
+#include <esp_err.h>
 
 #ifdef __cplusplus
 extern "C" {
```

winny2809, 2019-10-16:
Hello, I did the tests you asked me to do. Below is the result of the compilation, unfortunately no improvement.
The header of the i2cdev.h file is this:

#pragma once

#ifndef __I2CDEV_H__
#define __I2CDEV_H__

#ifdef __cplusplus
extern "C" {
#endif

#include <driver/i2c.h>

#include <freertos/FreeRTOS.h>

#include <freertos/semphr.h>

#include <esp_err.h>

Below is the result of the compilation:

> Executing task: C:\Users\Winny\.platformio\penv\Scripts\platformio.exe run <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30300.190916 (3.3.0), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <pcf8575>
|   |-- <i2cdev>
Compiling .pio\build\esp32doit-devkit-v1\src\main.o
In file included from lib/pcf8575/pcf8575.h:31:0,
                 from src\main.c:8:
lib/i2cdev/i2cdev.h:255:43: error: 'dev' undeclared here (not in a function)
         esp_err_t __ = i2c_dev_take_mutex(dev); \
                                           ^
lib/i2cdev/i2cdev.h:257:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:259:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:259:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:267:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:269:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:269:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:277:9: error: expected identifier or '(' before 'if'
         if (___ != ESP_OK) { \
         ^
lib/i2cdev/i2cdev.h:285:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
In file included from C:/Users/Winny/.platformio/packages/framework-espidf/components/esp32/include/esp_timer.h:44:0,
[... 174 lines cut ...]
     } while (0)
     ^
lib/i2cdev/i2cdev.h:259:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:267:9: error: expected identifier or '(' before 'if'
         if (__ != ESP_OK) return __;\
         ^
lib/i2cdev/i2cdev.h:269:5: error: expected identifier or '(' before '}' token
     } while (0)
     ^
lib/i2cdev/i2cdev.h:269:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
lib/i2cdev/i2cdev.h:277:9: error: expected identifier or '(' before 'if'
         if (___ != ESP_OK) { \
         ^
lib/i2cdev/i2cdev.h:285:7: error: expected identifier or '(' before 'while'
     } while (0)
       ^
*** [.pio\build\esp32doit-devkit-v1\lib588\pcf8575\pcf8575source.o] Error 1
==================================================== [FAILED] Took 26.84 seconds ====================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

UncleRus, 2019-10-16:
I'm at a loss. I've never dealt with the PlatformIO.
Could you try to manually build examples using `make` or `idf.py`?

winny2809, 2019-10-16:
Yes now i try it

UncleRus, 2019-10-16:
Since the assembly of examples with the recommended method was successful, I believe that you should ask questions related to the platformIO or VisualStudio on the esp-32 forum. I am closing this issue.

closed 2019-10-16 by UncleRus

daverathbone, 2022-02-26:
Would like to say that UncleRus could program the I2C with the paths he claims he has , then try it on another PC with the advice he  gives out , then he also would find it does not work!... it's not a native path in platformIO or Visual Studio plugin  tools. Or Eclipse...  Better I write one from scratch would have taken me less time...
What a pain..........

trombik, 2022-02-26:
we support esp-idf as a build system. I'm sorry to hear that your favourite tool doesn't work with the libraries. however, as the issue doesn't happen with the supported build system, I'd like to say that our efforts to help diagnosing the issue is only a best efforts basis. it would be faster to solve the issue if you ask someone who knows your tool than adding another "it doesn't work" comment without any details. thanks for your understanding.