UncleRus/esp-idf-lib #49: i2c driver install error

winny2809, 2019-10-18:
Hello everyone. 
I'm having trouble with the library for pcf8575.
After successfully compiling, the esp32 on serial gives me this output:

[1B][0;31mE (0) i2c: C:\Users\Winny\.platformio\packages\framework-espidf\components\driver\i2c.c:335 (i2c_driver_delete):i2c driver install error[1B][0m
[1B][0;31mE (10) I2C_DEV: Could not write to device [0x20 at 0]: 263[1B][0m

The hardware is properly installed, I also did other tests with another software is the device pcf8575 is normally seen at 0x20.

Any suggestions?

UncleRus, 2019-10-18:
Hi!
> (i2c_driver_delete):i2c driver install error
This is not an error, you can safely ignore this message.

UncleRus, 2019-10-18:
Error 263 (0x107) is ESP_ERR_TIMEOUT - no device with address 0x20 on I2C bus found.
Please check if your PCF8575 address is 0x20, if it's not, then change example source accordingly.

winny2809, 2019-10-18:
I tried with i2c Scan and the correct address is 0x20 and reads correctly.  the problem is something else

winny2809, 2019-10-21:
if write:

printf("i2c scanner\r\n\r\n");
// configure the i2c controller 0 in master mode, normal speed

	i2c_config_t conf;

	conf.mode = I2C_MODE_MASTER;

	conf.sda_io_num = 18;

	conf.scl_io_num = 19;

	conf.sda_pullup_en = GPIO_PULLUP_ENABLE;

	conf.scl_pullup_en = GPIO_PULLUP_ENABLE;

	conf.master.clk_speed = 100000;

	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &conf));

	printf("- i2c controller configured\r\n");

	

	// install the driver

	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0));

	printf("- i2c driver installed\r\n\r\n");

	

	printf("scanning the bus...\r\n\r\n");

	int devices_found = 0;

	

	for(int address = 1; address < 127; address++) {

	

		// create and execute the command link

		i2c_cmd_handle_t cmd = i2c_cmd_link_create();

		i2c_master_start(cmd);

		i2c_master_write_byte(cmd, (address << 1) | I2C_MASTER_WRITE, true);

		i2c_master_stop(cmd);

		if(i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS) == ESP_OK) {

			printf("-> found device with address 0x%02x\r\n", address);

			devices_found++;

		}

		i2c_cmd_link_delete(cmd);

	}

	if(devices_found == 0) printf("\r\n-> no devices found\r\n");

	printf("\r\n...scan completed!\r\n");

i found 0x20 adress

Zwirbeljupp, 2020-03-12:
I was running into exactly the same issue (ESP_ERR_TIMEOUT) when using the i2cdev.c code unchanged. When I was using an I2C scanner I was able to find all devices without issues.

Further analyzing the code it turned out that in i2c_setup_port(..) the driver is first installed before the parameters are set. After changing the code sequence to the one below I was able to communicate with the devices fine and did not get the timout errors anymore. 
@winny2809  : can you please confirm that this also solves your issues?

```
static esp_err_t i2c_setup_port(i2c_port_t port, const i2c_config_t *cfg)
{
    if (!cfg) return ESP_ERR_INVALID_ARG;

    esp_err_t res;
    if (!cfg_equal(cfg, &configs[port]))
    {
        ESP_LOGD(TAG, "Reconfiguring I2C driver on port %d", port);
        i2c_config_t temp;
        memcpy(&temp, cfg, sizeof(i2c_config_t));
        temp.mode = I2C_MODE_MASTER;

        // Daniel: moved this section before driver install!
        if ((res = i2c_param_config(port, &temp)) != ESP_OK)
            return res;

        if (installed)
            i2c_driver_delete(port);
#if HELPER_TARGET_IS_ESP32
        if ((res = i2c_driver_install(port, temp.mode, 0, 0, 0)) != ESP_OK)
            return res;
#elif HELPER_TARGET_IS_ESP8266
        if ((res = i2c_driver_install(port, temp.mode)) != ESP_OK)
            return res;
#endif

        installed = true;

        memcpy(&configs[port], &temp, sizeof(i2c_config_t));
        ESP_LOGD(TAG, "I2C driver successfully reconfigured on port %d", port);
    }

    return ESP_OK;
}
```

dougdev, 2020-04-16:
I have a BME280 connected to a Wemos D1 Mini over i2c. I am using the latest versions of both ESP-IDF and ESP_Components_Library. My task for reading the BME280 is based off the example code and was crashing during initialization in the bmp280_init() function. I tried using the i2c scanner code provided by @winny2809 above and found it too crashed. However, when I moved the i2c_driver_install() call to be before the call to i2c_param_config(), it ran just fine and located the BME280. On a hunch, I reverted the change made to i2c_setup_port() in the Mar 14 commit above so that the driver install was done first and my code now works as expected. Not sure if this is an ESP8266 vs ESP32 thing.

UncleRus, 2020-04-16:
So, we have two completely different behaviors: for someone, an error occurs if the call to i2c_param_config() occurs before i2c_driver_install(), and for someone the other way round.
It seems that I will need to check all the options on different versions of ESP-IDF for both ES32 and ES8266.
Hmm...

dougdev, 2020-04-16:
I need to make a correction to my earlier post. Since I"m writing for the Wemos, I'm using the latest version of ESP8266_RTOS_SDK, not ESP-IDF. The ESP-IDF Programming Guide ([https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/peripherals/i2c.html](url)) does explicitly state that the i2c_param_config() needs to come before the i2c_driver_install(), so I"m thinking this problem might just be due to the current state of Espressif's refactoring of ESP8266_RTOS_SDK to the IDF style.

UncleRus, 2020-04-16:
9b26a6f Ok, now it's `i2c_param_config() -> i2c_driver_install()` for ESP-IDF and `i2c_driver_install() -> i2c_param_config()` for ESP8266 RTOS SDK, because of:

https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-reference/peripherals/i2c.html
> i2c_param_config() must be used after calling i2c_driver_install()

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html
> After the I2C driver is configured, install it by calling the function i2c_driver_install()

UncleRus, 2020-04-20:
I'm closing this issue, feel free to reopen it.

closed 2020-04-20 by UncleRus

jaami, 2024-07-27:
> Hi!
> 
> > (i2c_driver_delete):i2c driver install error
> 
> This is not an error, you can safely ignore this message.

I am new to IOT and recently got involved in ESP32 app development. Using esp-idf and i doubt if im using the latest updates. anyway I got rid of this annoying error which actually seems not an error.

The function ds1307_init_desc() from ds1307.c gives call to
function i2c_driver_install from i2c.c and when driver is already
installed it returns error in else part.
The error in else part was just converted to ESP_OK.
Otherwise error will show "i2c driver install error".
That is not error so should not be displayed.