UncleRus/esp-idf-lib #50: E (2387) DHTxx: Initialization error, problem in phase 'C'

Somusundram, 2019-10-19:
**Device type**
- [] ESP32
- [X] ESP8226

**Framework version**
- [] ESP-IDF latest master
- [] ESP-IDF 3.2.2
- [] ESP8266 RTOS SDK latest master
- [X] ESP8266 RTOS SDK 3.2

Hi,

I tried your DHT (i use DHT11) library and seems it stuck in this phase:

E (2387) DHTxx: Initialization error, problem in phase 'C'

I use GPIO 14 with 4.7k pull up resistor.

afcec, 2019-10-28:
Same situation with ESP32, problem in phase 'C', GPIO 17 and PCB including pull up resistor...

closed 2020-04-20 by UncleRus

LaurenceGA, 2021-05-06:
@Somusundram @afcec did either of you figure out what was wrong?

afcec, 2021-05-06:
No @LaurenceGA , sorry, I did some more tests and after that I just discarded DHT and took a different approach...

LaurenceGA, 2021-05-07:
> No @LaurenceGA , sorry, I did some more tests and after that I just discarded DHT and took a different approach...

Oh, sorry to hear.
Actually, in my case it was just some faulty breadboard wiring...

Prabhuelectro, 2021-09-04:
@LaurenceGA what did you make faulty breadboard wiring?

LaurenceGA, 2021-09-04:
> @LaurenceGA what did you make faulty breadboard wiring?

I don't understand the question. Things just weren't plugged in properly. When I re-did the connections, it worked.

Prabhuelectro, 2021-09-04:
@LaurenceGA thankyou same happened here, I have connected the data pin to the wrong pin.

colesnicov, 2023-07-06:
I have a similar problem only in **phase 'B'**. The strangest thing is that sometimes it works and sometimes it has a problem with the **phase 'B'**.

UncleRus, 2023-07-06:
https://github.com/UncleRus/esp-idf-lib/blob/master/FAQ.md#my-dht-sensor-doesnt-work-welldoesnt-work-at-all

colesnicov, 2023-07-07:
So. 5 Volts is the right way. 
But, in the future I will not have a 5 Volt power supply on the board. Although I learned that the DHT11 has a power supply range of 3.5 to 5 volts. **AM2301** may be needed as an alternative to DHT11. Do you have experience with it, does it work from a 3.3 volt supply?

UncleRus, 2023-07-07:
No, I did not use AM2301.
Regarding the supply voltage: it is necessary to take into account the voltage drop on the wires and measure the supply voltage at the sensor itself. This is 3.3V on the board, but less on the sensor.

UncleRus, 2023-07-07:
And please, ask further questions not in a necropost in a closed issue, but in "Discussions".

espark-Tayyab, 2023-11-15:
I have the same problem