UncleRus/esp-idf-lib #54: Got some errors on ESP32

MusabAybek, 2019-11-15:
**Devcie type**
- [x] ESP32

**Framework version**
- [x] ESP-IDF v3.3

**Description of the bug**
Hi. Firstly I have to say that I'm new on Esp32 world. I've tried to flash the example code of DS1307. And got those lines of messages as follows repeatedly. What is the reason could be? I'll be very much appreciated if somebody can help...

The messages I got  after I make monitor;

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6344
load:0x40078000,len:11624
load:0x40080400,len:6648
entry 0x40080764
I (29) boot: ESP-IDF v3.3 2nd stage bootloader
I (29) boot: compile time 23:39:26
I (29) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0af40 ( 44
864) map
I (108) esp_image: segment 1: paddr=0x0001af68 vaddr=0x3ffb0000 size=0x01ed8 (
7896) load
I (111) esp_image: segment 2: paddr=0x0001ce48 vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Musab/esp/esp-idf/components/free
rtos/xtensa_vectors.S:1779

I (115) esp_image: segment 3: paddr=0x0001d250 vaddr=0x40080400 size=0x02dc0 ( 1
1712) load
I (128) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x15054 ( 8
6100) map
0x400d0018: _flash_cache_start at ??:?

I (163) esp_image: segment 5: paddr=0x00035074 vaddr=0x400831c0 size=0x062bc ( 2
5276) load
0x400831c0: spi_flash_mmap at C:/msys32/home/Musab/esp/esp-idf/components/spi_fl
ash/flash_mmap.c:320

I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (180) cpu_start: Pro cpu up.
I (183) cpu_start: Application information:
I (188) cpu_start: Project name:     example-ds1307
I (194) cpu_start: App version:      1
I (198) cpu_start: Compile time:     Nov 15 2019 23:39:55
I (204) cpu_start: ELF file SHA256:  23b3a89ef017dc9a...
I (210) cpu_start: ESP-IDF:          v3.3
I (215) cpu_start: Starting app cpu, entry point is 0x40080f54
0x40080f54: call_start_cpu1 at C:/msys32/home/Musab/esp/esp-idf/components/esp32
/cpu_start.c:270

I (207) cpu_start: App cpu up.
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (238) heap_init: At 3FFB2F38 len 0002D0C8 (180 KiB): DRAM
I (245) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (251) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (257) heap_init: At 4008947C len 00016B84 (90 KiB): IRAM
I (263) cpu_start: Pro cpu start user code
I (282) cpu_start: Chip Revision: 1
W (282) cpu_start: Chip revision is higher than the one configured in menuconfig
. Suggest to upgrade it.
I (285) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
**E (0) i2c: C:/msys32/home/Musab/esp/esp-idf/components/driver/i2c.c:335 (i2c_dri
ver_delete):i2c driver install error
E (10) I2C_DEV: Could not write to device [0x68 at 0]: -1**
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085290
0x40085290: _esp_error_check_failed at C:/msys32/home/Musab/esp/esp-idf/componen
ts/esp32/panic.c:720

file: "C:/msys32/home/Musab/esp/workspace/rtcson/examples/ds1307/main/main.c" li
ne 31
func: ds1307_test
expression: ds1307_set_time(&dev, &time)

ELF file SHA256: 23b3a89ef017dc9aee44448e2acb362b2d5ea0517fdf868f1fe1d87ed90bf3f
1

Backtrace: 0x40084e2c:0x3ffb8080 0x40085293:0x3ffb80a0 0x400d24ae:0x3ffb80c0 0x4
00878c1:0x3ffb8140
0x40084e2c: invoke_abort at C:/msys32/home/Musab/esp/esp-idf/components/esp32/pa
nic.c:715

0x40085293: _esp_error_check_failed at C:/msys32/home/Musab/esp/esp-idf/componen
ts/esp32/panic.c:721

0x400d24ae: ds1307_test at C:/msys32/home/Musab/esp/workspace/rtcson/examples/ds
1307/main/main.c:31 (discriminator 1)

0x400878c1: vPortTaskWrapper at C:/msys32/home/Musab/esp/esp-idf/components/free
rtos/port.c:403

MusabAybek, 2019-11-15:
That group of message that I mentioned above, comes again and again and the time is always the same; 
ets Jun  8 2016 00:22:57

closed 2019-11-16 by MusabAybek

ChristiaanHaas, 2020-11-09:
Hey Musab! I'm receiving the same errors now, how did you manage to fix it in the end?