UncleRus/esp-idf-lib #55: DS18X20 abort() was called at PC 0x4008267b

dominico2000, 2019-11-23:
**Devcie type**
- [x] ESP32

**Framework version**
- [x] ESP-IDF latest master

**Describe the bug**
When I try read temperature from ds18b20 sensor I get maximally one read (not correct) and next I get error like this
`abort() was called at PC 0x4008267b on core 1
0x4008267b: lock_acquire_generic at /home/dominik/esp32/esp-idf/components/newlib/locks.c:143

ELF file SHA256: 1f7ff62be29ee34dc166c893d75f93478221a9afea43bbb21f3208096a0dbef4

Backtrace: 0x40084a45:0x3ffb9460 0x40084dc9:0x3ffb9480 0x4008267b:0x3ffb94a0 0x400827a5:0x3ffb94d0 0x400dce1a:0x3ffb94f0 0x400dfe29:0x3ffb9800 0x40081db5:0x3ffb9830 0x400d4b21:0x3ffb9880 0x400d3e23:0x3ffb98e0 0x400d40e4:0x3ffb9920 0x400d3b98:0x3ffb9940 0x400d3dd1:0x3ffb9960 0x400d3ab8:0x3ffb9980 0x40086fe5:0x3ffb9a20
0x40084a45: invoke_abort at /home/dominik/esp32/esp-idf/components/esp32/panic.c:157

0x40084dc9: abort at /home/dominik/esp32/esp-idf/components/esp32/panic.c:174

0x4008267b: lock_acquire_generic at /home/dominik/esp32/esp-idf/components/newlib/locks.c:143

0x400827a5: _lock_acquire_recursive at /home/dominik/esp32/esp-idf/components/newlib/locks.c:171

0x400dce1a: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x400dfe29: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34

0x40081db5: esp_log_write at /home/dominik/esp32/esp-idf/components/log/log.c:219

0x400d4b21: gpio_config at /home/dominik/esp32/esp-idf/components/driver/gpio.c:354 (discriminator 9)

0x400d3e23: setup_pin at /home/dominik/esp32/esp-idf-lib/components/onewire/onewire.c:69 (discriminator 4)

0x400d40e4: onewire_power at /home/dominik/esp32/esp-idf-lib/components/onewire/onewire.c:228

0x400d3b98: ds18x20_measure at /home/dominik/esp32/esp-idf-lib/components/ds18x20/ds18x20.c:65

0x400d3dd1: ds18x20_measure_and_read_multi at /home/dominik/esp32/esp-idf-lib/components/ds18x20/ds18x20.c:142

0x400d3ab8: ds18x20_test at /run/media/dominik/Data2/Programming/ESP32/ds18b20_and_dht_sensors/build/../main/main.c:54

0x40086fe5: vPortTaskWrapper at /home/dominik/esp32/esp-idf/components/freertos/port.c:143
`

alafi, 2019-11-25:
I am having the same issue. The root cause of this issue is due to `ESP_LOGX()` functions being called by `gpio_config()` function inside a critical section which is not supposed to be done like that.

Check a similar issue here https://github.com/espressif/esp-idf/issues/3009 (which was resolved obviously).

I don't really know how to fix this one (besides commenting out the critical section macros in `ds18x20_measure` which is not a proper solution), but waiting for @UncleRus to comment on this.

alafi, 2019-11-25:
One workaround is to disable informative logging level for `gpio` library.

Invoke the following function in the beginning of your main method:
`esp_log_level_set("gpio", ESP_LOG_WARN);`

commit a08e8f82 (2019-11-25): Fix #55
closed 2019-11-25 by UncleRus