UncleRus/esp-idf-lib #58: Try to read temperature from single DS18b20

kusumy, 2019-12-05:
Hi, 
I have below code to read temperature from single DS18b20, but it always error.
Please help.

__________________________________________________________________________________________
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include "esp_log.h"
#include "esp_event_loop.h"
#include "esp_system.h"
#include "ds18x20.h"

void read_temp(void *pvParameter)
{
	
	ds18x20_addr_t addr = 0x280F090FE40F2233;
	float  temp = 0;
	esp_err_t err;
	while (1)
	{
		err = ds18x20_measure(19, addr, true);	
		err = ds18x20_read_temperature(19, addr, &temp);
		printf("Temperature is: %f\n", temp);
		
		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
}

void app_main()
{
	xTaskCreate(&read_temp, "read_temp", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
}

UncleRus, 2021-03-26:
See `ds18x20_single` example.
Note that ROM codes are reversed in the current `onewire` implemementation.

closed 2021-03-26 by UncleRus

zenz, 2021-11-01:
@UncleRus 
Hi, It seems the ds18x20 lib cannot read temperature while using esp8266_rtos_sdk v3.4
no matter ds18x20_multi or ds18x20_single gets empty value like below.
```
I (1019) ds18x20_test: Sensor ffffffffffffffff: °C
I (2769) ds18x20_test: Sensor ffffffffffffffff: °C
```

By the way. while using esp8266_rtos_sdk 3.4, the GPIO pin must be init before measure. Or else it cannot find sensors.

zenz, 2021-11-01:
OK, it's because the newlib defaults to nano mode, make menuconfig to swithc it to normal mode, then it comes back.