UncleRus/esp-idf-lib #59: Request driver for DS2438 (Battery Monitor)

kusumy, 2019-12-05:
Hi, 
I would like to request a driver for Dallas DS2438, please.
Need it so much for my school task.

Regards

Raghav3107, 2022-06-18:
Hello, @kusumy  @UncleRus 

I tried to port the driver code of Dallas DS243.
I do not have the device so not tested the code on the device if anyone has can test and update me.
It's building successfully on esp idf v4.4.1.

Thanks.

UncleRus, 2022-06-18:
Hi @Raghav3107 !
First of all, thank you.

1. The first and most important note: drivers must be stateless! If you need to save some state between calls, the device descriptor structure is used. Your library can only be used with _single_ device, because of the saving device GPIO and mode to internal `_pin` and `_mode` variables. The same goes for the `ds2438_update()` function: it should just read and return the voltage and temperature values without saving them. To avoid this, declare device descriptor struct and pass it by pointer to all functions. 
2. Please avoid `int` parameters when they can be replaced with enums.
3. Please add C++ guards in header.
4. Move internal definitions such as register numbers from the header to the source. They are only used internally and pollutes the global namespace in header.
5. Clean includes in header. Why is there `stdio.h` and `stdlib.h`? 
6. Avoid using of `func(void)` instead of `func()`.
7. Do not use tabs. Spaces only.

Raghav3107, 2022-06-18:
Hello @UncleRus

Thanks for the advice and guidance. 

The first time, I m trying to contribute to the open-source repo, I m not well known about coding standards and other things.
I will update lib as per the coding standard and your instructions.

Thanks!

Raghav3107, 2022-06-19:
Hello @UncleRus.

I updated the repository as per your instruction. 
Please check it once again.

Thanks!

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/core/issues/34