UncleRus/esp-idf-lib #60: multiple i2c bus error

steeveone, 2020-01-02:
**Devcie type**
- [ ] ESP32
- [x] ESP8226

**Framework version**
- [x] ESP-IDF latest master
- [ ] ESP-IDF 3.2.2
- [ ] ESP8266 RTOS SDK latest master
- [ ] ESP8266 RTOS SDK 3.2

I have "merged" examples for BH1750 and BMP280 sensors, defining different GPIO pins for SDAs and SCLs (so i have 2 buses). 
Tested separately both buses works (not initing one or the other sensor and relative task), if I configure both sensors i get

E (586) i2c: i2c driver install error

E (591) BMP280: Sensor not found
and a panic:
0x40221e30: _esp_error_check_failed at /home/steeve/newesp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/panic.c:173

file: "sensors.c" line 27
func: bmp280_task
expression: bmp280_init(&dev, &params)
abort() was called at PC 0x40221e33 on core 0

Can I2C driver work on 2 buses simultaneously?

UncleRus, 2020-01-03:
Latest ESP8266 RTOS SDK supports only one I2C port
> components/esp8266/include/driver/i2c.h
```C
typedef enum {
    I2C_NUM_0 = 0,  /*!< I2C port 0 */
    I2C_NUM_MAX
} i2c_port_t;
```

UncleRus, 2020-01-03:
You can use your devices like this:
```C
static i2c_dev_t bh1750;
static bmp180_dev_t bmp180;
...
void some_task(void *pvParameters)
{
    // Use I2C devices
   ....
}
...
void other_task(void *pvParameters)
{
    // Use I2C devices
   ....
}

void app_main()
{
    ....
    ESP_ERROR_CHECK(i2cdev_init()); // Init library

    memset(&bh1750, 0, sizeof(i2c_dev_t)); // Zero BH1750 descriptor
    memset(&bmp180, 0, sizeof(bmp180_dev_t)); // Zero BMP180 descriptor

    // Init descriptors
    ESP_ERROR_CHECK(bh1750_init_desc(&bh1750, BH1750_ADDR, 0, BH1750_SDA_GPIO, BH1750_SCL_GPIO));
    ESP_ERROR_CHECK(bmp180_init_desc(&bmp180, 0, BMP180_SDA_GPIO, BMP180_SCL_GPIO));
    ...
    // Init devices
    ESP_ERROR_CHECK(bmp180_init(&bmp180));
    ESP_ERROR_CHECK(bh1750_setup(&bh1750, BH1750_MODE_CONTINIOUS, BH1750_RES_HIGH));
    ...
    // Create tasks and so on
    ...
}
```

UncleRus, 2020-01-06:
I'm closing this issue for now, feel free to reopen.

closed 2020-01-06 by UncleRus