UncleRus/esp-idf-lib #61: Library imported correctly but esp32 is restarting.

ravishree19, 2020-01-09:
**Devcie type**
- ESP32
**Framework version**
-  ESP-IDF latest master
I have to interface dht22 sensor in ESP32 using esp-idf.I refered https://github.com/UncleRus/esp-idf-lib for dht22 and i was unable to include the dht lib.Can you please help me with the CmakeList.txt file and the Make file to include path.The issue after following the procedure in above link is that esp32 is restarting.I didnt know how CmakeList file works.
Screenshot of the errors after flashing the code in it.
[image]
[image]
[image]

XxKavosxX, 2020-04-17:
Yes, the same to me

zmnv, 2020-05-27:
Have the same trouble.
I use 'dht' from esp-idf-lib last master from now.
Also 'esp-idf' last for now.
Trying to use DH11 (DHT_TYPE_DHT11 and others). Also my sensor is not connected))

JoaoGarrido, 2020-06-08:
Had the same issue and solved it doing the following:
On the dht.c file do the following change:
```c
#define CHECK_LOGE(x, msg, ...) do { \
        esp_err_t __; \
        if ((__ = x) != ESP_OK) { \
            PORT_EXIT_CRITICAL; \
            ESP_LOGE(TAG, msg, ## __VA_ARGS__); \
            return __; \
        } \
    } while (0)
```

closed 2020-07-01 by UncleRus