UncleRus/esp-idf-lib #62: ds18 sensor not detected!!

ravishree19, 2020-01-10:
i am using esp-idf version 4.1 and the ds18 example code is correctly complied but after flashing on esp32 and in monitor it is only showing "sensor not detected".I have connected ds18 on gpio17 with pullup resistor 4.7k in between vcc and data pin.  
[image]

DheeranYuvaraj, 2020-01-21:
Dear @UncleRus 

Any update on this?

UncleRus, 2020-01-25:
This is output from examples/ds18x20 with the latest esp-idf master:
(ESP-32, internal pullup resistor)
```
1 sensors detected:
  Sensor b203168224a8ff28 reports 26.000000 deg C (78.800003 deg F)

  Sensor b203168224a8ff28 reports 26.375000 deg C (79.474998 deg F)

  Sensor b203168224a8ff28 reports 26.812500 deg C (80.262497 deg F)

  Sensor b203168224a8ff28 reports 27.250000 deg C (81.050003 deg F)

  Sensor b203168224a8ff28 reports 27.625000 deg C (81.724998 deg F)

  Sensor b203168224a8ff28 reports 27.875000 deg C (82.175003 deg F)

  Sensor b203168224a8ff28 reports 28.062500 deg C (82.512497 deg F)

```
Looks like you have power/connectivity problem.

closed 2020-01-29 by UncleRus

eletors, 2024-10-23:
I had a similar problem on esp-idf 5.3.1 I solved it by swapping:
    
    gpio_set_pull_mode(SENSOR_GPIO, GPIO_PULLUP_ONLY);
    
to:
     
    gpio_config_t io_conf;
    io_conf.mode = GPIO_MODE_INPUT_OUTPUT_OD;
    io_conf.pin_bit_mask = 
        GPIO_BIT(SENSOR_GPIO)
        ;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 1;
    ESP_ERROR_CHECK(gpio_config(&io_conf));