UncleRus/esp-idf-lib #66: CCS811B

digimatt22, 2020-01-13:
Is anyone working on porting the CCS811B driver over from esp-open-rtos?  If not I may attempt in the near future and submit a PR as I will need it in an upcoming project. Great library!

UncleRus, 2020-01-14:
> I may attempt in the near future and submit a PR
That's would be great!

UncleRus, 2020-01-26:
I've ported this driver, see #67
Could you test it?

digimatt22, 2020-01-26:
Absolutely but it will be a few day as I am traveling to a conference.  Will have access to the hardware again on Friday.

digimatt22, 2020-02-01:
I have begun testing the driver and found the following items.   I will continue to work through them but wanted to give an update.  

ccs811_init_desc -> port and addr are flipped compared to other libraries.  I have changed their order to stay consistent with other implementations

ccs811_init_sensor -> within this function there is a write of no data to a specific register (CCS811_REG_APP_START) the i2c_dev_write_reg does not allow for this.  I can disable this check but all future calls appear to fail once I have.  I will continue to look into this, if you have seen this before and have devised a solution please let me know.

UncleRus, 2020-02-02:
I updated PR according to your comments

UncleRus, 2020-02-05:
@digimatt22 Any news on testing?

digimatt22, 2020-02-09:
Testing is paused right now as I wait for a new logic analyzer to arrive so I can see what’s going on vs guess. It arrives later this week.

UncleRus, 2020-12-13:
I'm going to merge this driver as it is in humble hope that someone will someday try to use it and be kind enough to create an issue in this repository if something goes wrong.

trombik, 2020-12-13:
I'm fine with it. I ordered a breakout module a week ago, to be delivered in another week or two.

closed 2021-03-12 by UncleRus