UncleRus/esp-idf-lib #68: BME280 Error: Could not read from device [0x77 at 0]: 263

Djaison, 2020-02-25:
**Devcie type**
- [x] ESP32

**Framework version**
- [x] ESP-IDF latest master

**Using Library**
- [x] bmp280

Hello! I am using a library bmp280, sensor bme280. I compile the program from the examples folder. I use GPIOs 21,22 with 4.7k pull up resistor. After power up, I see in serial monitor the following:

E (4000) I2C_DEV: Could not read from device [0x77 at 0]: 263
E (4000) BMP280: Sensor not found
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x40086304

This sensor works great with other libraries in the Arduino environment. I compared the signals in both cases with the help of logic analyzer. On this image I see that the sensor responds to the read request BMP280_REG_ID register, but the program does not see this, and driver wait 1000ms timeout.
[image]
What could cause this problem? Frequency clock I decreased to 100kHz.
Please, help me!

Djaison, 2020-02-25:
In the picture below, signals received with Arduino library and work fine
[image]

UncleRus, 2020-02-26:
This is definitely a bug. Thanks for the report!

UncleRus, 2020-02-29:
Strange, I just tested the BMP280 example and everything worked fine:
```
BMP280: found BMP280
Pressure: 97365.36 Pa, Temperature: 24.69 C
Pressure: 97365.03 Pa, Temperature: 24.70 C
Pressure: 97367.66 Pa, Temperature: 24.70 C
Pressure: 97364.78 Pa, Temperature: 24.70 C
Pressure: 97367.73 Pa, Temperature: 24.71 C
```
Unfortunately, I do not have bme280 at hand for the test.

UncleRus, 2020-04-01:
@Djaison Could you test this again with latest esp-idf-lib master? It looks like the 1d8ea2d fixes this bug.

Djaison, 2020-04-24:
@Djaison It is working! Thank you!

closed 2020-04-24 by Djaison