UncleRus/esp-idf-lib #72: HX711 ESP-IDF v4.0 version error

Squidyman, 2020-04-03:
**Devcie type**
ESP32

**Framework version**
ESP-IDF latest master 4.0

**Describe the bug**
The HX711 driver does not work with IDF 4.0

In file included from C:/Users/mattb/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
../../../../components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
../../../../components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired,  #order)

This is based off of the same error and proposed solution from this issue https://github.com/OLIMEX/ESP32-POE/issues/8

UncleRus, 2020-04-03:
It seems that this error may be due to an incompatible version of the esp32 toolchain or sdkconfig of the project.
Try to:
- Install esp-idf with toolchain according to the instructions 
  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html
- Delete the build directory with the sdkconfig file.
- Create the default configuration using make defconfig and rebuild the project.

Please report the results.

Squidyman, 2020-04-05:
> It seems that this error may be due to an incompatible version of the esp32 toolchain or sdkconfig of the project.
> Try to:
>
> * Install esp-idf with toolchain according to the instructions [...]

Will do, it compiles on my laptop which I haven't updated in a while, however on my desktop with a new and latest install I'm getting errors (even according to espressif instructions). I'll report back with findings and report from your suggestion.

UncleRus, 2020-06-18:
Feel free to reopen

closed 2020-06-18 by UncleRus