UncleRus/esp-idf-lib #73: Created a wifi logger component for esp-idf, can be included in the components?

VedantParanjape, 2020-04-14:
**Devcie type**
- [x] ESP32

**Framework version**
- [x] ESP-IDF latest master

I developed a esp-idf component which sends esp generated logs over to a PC through wifi. It can send through UDP, TCP and websocket. It's user selectable. If it's possible can it be included in your repo?

Link: https://github.com/VedantParanjape/esp-wifi-logger

Regards, Vedant

UncleRus, 2020-04-14:
Personally, I like this. But I would like to know the opinion of @trombik.

trombik, 2020-04-14:
it is technically possible. but, why? would you explain the benefit? your library has its own repository, and your users can install it. what are you trying to solve?

VedantParanjape, 2020-04-15:
Hey

I just want to increase visibility, get people to use and contribute it It's okay if not.

Regards, Vedant

trombik, 2020-04-15:
I found this repository useful because it has many drivers for devices. a device driver, usually, doesn't require maintenance when the implementation gets mature because chip makers don't change their products often. on the other hand, protocols keep changing. new requirements arise, users ask new features, etc. it would be better for you to keep your repository.

I actually have a local branch in my fork to build curl. I gave up to merge the branch because it is too much work to maintain. if I ever publish my work, it would be its own repository, not in this repository.

as stated in README, we are trying to support ESP8266 and ESP32. but unfortunately, ESP8266 doesn't have same libraries in the base framework, often lagged behind. MQTT is a good example. it would be very hard for you to support ESP8266.

I do not own the repository. this is just my opinion.

UncleRus, 2020-04-15:
Sakurai has the same rights in this repository as mine and his opinion has the same weight as mine. I regret to inform you that I am closing this issue.

closed 2020-04-15 by UncleRus

VedantParanjape, 2020-04-15:
@trombik I respect your opinion, it's sensible. I agree with the points.

Regards, Vedant