UncleRus/esp-idf-lib #79: Not able to display text properly over MAX7219 LED

Ritesh236, 2020-04-26:
Hello,

We are planning to use MAX7219 Total 3 LEDs with ESP32 Dev Kit C V4 with cascaded method. 

We have ported all files from this component and created one task with below example shown.

`/******************************************************************************
* FunctionName : max7129_task
* Description  : This function will Configure SPI bus with MAX7129
* Parameters   : none
* Returns      : none
*******************************************************************************/
void max7129_task()
{
        esp_err_t ret;

        // Configure SPI bus
        spi_bus_config_t cfg = {
                        .mosi_io_num = PIN_NUM_MOSI,
                        .miso_io_num = -1,
                        .sclk_io_num = CLK_PIN,
                        .quadwp_io_num = -1,
                        .quadhd_io_num = -1,
                        .max_transfer_sz = 0,
                        .flags = 0
        };
        ret = spi_bus_initialize(HOST, &cfg, 1);
        ESP_ERROR_CHECK(ret);
        // Configure device
        max7219_t dev = { .cascade_size = 3, .digits = 8, .mirrored = true };

        ret = max7219_init_desc(&dev, HOST, CS_PIN);
        ESP_ERROR_CHECK(ret);

        char buf[9];

        ret = max7219_init(&dev);
        ESP_ERROR_CHECK(ret);

        while (1) {
                DEBUG_LOGI(UART, TAG, "Display STRING\n");

                max7219_clear(&dev);
                max7219_draw_text_7seg(&dev, 0, "7219TEST");
                vTaskDelay(DELAY / portTICK_PERIOD_MS);

//                      max7219_clear(&dev);
//                      sprintf(buf, "%2.4f A", 34.6782);
//                      max7219_draw_text_7seg(&dev, 0, buf);
//                      vTaskDelay(DELAY / portTICK_PERIOD_MS);

                DEBUG_LOGI(UART, TAG, "Display DIGIT\n");
                max7219_clear(&dev);
                sprintf(buf, "%08x", 12345678);
                max7219_draw_text_7seg(&dev, 0, buf);
                vTaskDelay(DELAY / portTICK_PERIOD_MS);
        }
}`

We have compiled it with ESP32 IDF 3.2 and flashed into development kit but not getting proper scrolling string like we got using Arduino example over same board.

I have uploaded sample video with 3 LEDs for reference purpose in which you can see issue which we are facing. 

Let me know if need anything else from my end to resolve issue.

Regards,
Ritesh Prajapati
[VID_20200424_181201_ESP32_IDF.zip](https://github.com/UncleRus/esp-idf-lib/files/4534969/VID_20200424_181201_ESP32_IDF.zip)

Ritesh236, 2020-04-26:
Hello,

Also let us know can it be possible to scroll text from left to right or right to left if we connected more than one LED?

Regards,
Ritesh Prajapati

UncleRus, 2020-04-26:
Looks like 
```C
max7219_t dev = { .cascade_size = 3, .digits = 8, .mirrored = true };
```
is wrong. I suggest yo to do something like
```C
max7219_t dev = { .cascade_size = 3, .digits = 24 /* cascade_size * digits in one cascade */, .mirrored = true };
```

Ritesh236, 2020-04-26:
@UncleRus, 

Thanks for your quick response. I think I have checked with above but still will confirm it again. 

Also I would like to know that which API can be useful to scroll text continuously as we are using total 3 cascaded LEDs together? Because we want to continuously display text with some delay intervals. 

Hope you can understand my question for that. 

Regards, 
Ritesh Prajapati

Ritesh236, 2020-04-27:
@UncleRus,

We have tried with below settings but still no any success for that. only single LED 8X8 is working but characters are not properly displayed

max7219_t dev = { .cascade_size = 3, .digits = 24 /* cascade_size * digits in one cascade */, .mirrored = true };

Are you sure that it will be worked with multiple cascaded LEDs like 3 into our case?

Let me know if any other resolutions for same.

Regards,
Ritesh Prajapati

Ritesh236, 2020-04-27:
@UncleRus , 

We have checked with your solution but it didn't work well and only displayed into one LED but still not proper. 

So what we can do to resolve it as soon as possible because we need to fix it urgently.. 

Regards, 
Ritesh Prajapati

closed 2020-07-01 by UncleRus