UncleRus/esp-idf-lib #81: SHT31 IDF 4.1 error

jmpmscorp, 2020-05-06:
**Device type**
- [x] ESP32 (esp32-devkitc-v4)

**Framework version**
- [x] ESP-IDF release/4.1 branch (ESP-IDF v4.1-beta1-182-g9778b163b)

**Describe the bug**
I have been testing sht3x component example. Program always fail and reset at sht3x_init(&dev), and the error shown is:
```
 E (315) I2C_DEV: Could not write to device [0x44 at 0]: -1
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400861f0
0x400861f0: _esp_error_check_failed at D:/esp-idf_4.1/components/esp32/panic.c:739

file: "../main/main.c" line 145
func: app_main
expression: sht3x_init(&dev)

ELF file SHA256: d99006b6ec61c296

Backtrace: 0x40085c45:0x3ffb4830 0x400861f3:0x3ffb4850 0x400d3dbd:0x3ffb4870 0x400d0d0a:0x3ffb48a0 0x400884d1:0x3ffb48c0
0x40085c45: invoke_abort at D:/esp-idf_4.1/components/esp32/panic.c:157

0x400861f3: _esp_error_check_failed at D:/esp-idf_4.1/components/esp32/panic.c:740

0x400d3dbd: app_main at E:\JM\Code\esp-idf\sht31_test\build/../main/main.c:145 (discriminator 1)

0x400d0d0a: main_task at D:/esp-idf_4.1/components/esp32/cpu_start.c:564

0x400884d1: vPortTaskWrapper at D:/esp-idf_4.1/components/freertos/port.c:143

Rebooting...
```

I have tried many pin combinations, downgrade I2C_FREQ_HZ at source component to 100kHz and 400kHz but error always appears.

I have tested sensor in another Arduino board (1284p MCU) with Arduino framework and sensor measurement works perfectly. Also, I have tried same sensor with same esp-devkitc board using Arduino framework instead of IDF framework and board and sensor work great too.

Some days ago, I noticed similar issue with bme280. Reading issues, I found that we should update lib to last version and, done it, bme280 worked again. However, sht31 seems, at least mine, not to work yet.

Could you give me any tip?

Thanks in advance.

UncleRus, 2020-05-07:
Looks like a driver bug.
I do not have SHT31 on hand and I will be extremely grateful if you analyze the data exchange between the ESP32 and the sensor using an oscilloscope or logic analyzer for this library and for arduino library.

jmpmscorp, 2020-05-07:
Thanks for reply @UncleRus .

Of course, here you have many screenshots. Feel free to delete anyone if you consider there are too much pictures.

Process in Arduino sketch is SoftReset - Configure Measure High Rep and Read values. In IDF, it fails at start SoftReset.

**Arduino 1824p board & Arduino framework:**

Software Reset:
[image]

Measure High Rep:
[image]

Values:
[image]

**esp32-devkitc-v4 & Arduino framework:**

Software Reset:
[image]

Measure High Rep:
[image]

Values:
[image]

**esp32-devkitc-v4 & IDF framework:**

1MHz I2C freq, SoftReset:
[image]

100kHz I2C freq, SoftReset:
[image]

IDF error seems to be related with last ACK. Screenshots show always a NACK in last byte.

UncleRus, 2020-05-10:
ESP-IDF has problems with I2C clock stretching.
Try to add to your code  `i2c_set_timeout(<I2C port number>, 0xffff);` before `sht3x_init()`
And if this works I'll implement it in i2cdev.

jmpmscorp, 2020-05-11:
I have done it but nothing changes. Same error at same line.

UncleRus, 2020-05-12:
Could you try [i2c_clk_stretch](https://github.com/UncleRus/esp-idf-lib/tree/i2c_clk_stretch) branch?

jmpmscorp, 2020-05-12:
Hi again.

I've just check new branch and same result but... it seems error is not related with clock stretching. I explain.

My fault was that I didn't test CONFIG_SHT3X_DEMO_SSHL and CONFIG_SHT3X_DEMO_SSLL. Like I detected error on sht3x_init, I thougth error was there. After test new branch, as said, I have noticed same error, so I decided define CONFIG_SHT3X_DEMO_SSHL, build and flash. My suprise, It works. Same result if defined CONFIG_SHT3X_DEMO_SSLL. Then, I returned back and delete "define", so CONFIG_SHT3X_DEMO_PERIODIC have been build and flashed again. 

After that, terminal monitor showed first measurement good but next secuences, new error. This is the log:

```
SHT3x Sensor: 24.74 °C, 48.87 %
E (11130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (16130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (21130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (26130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (31130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (36130) SHT3x: Measurement is not started
Could not get results: 259 (ESP_ERR_INVALID_STATE)E (41130) SHT3x: Measurement is not started
```

If I restart with onboard reset button, error ```E (315) I2C_DEV: Could not write to device [0x44 at 0]: -1``` is shown and sht3x_init failed everytime. However, if I unplug sensor from power source and replug, same log before is shown... first measurement is good and next "could not get result..."

Then, maybe there is a bug in fuctions used by DEMO_PERIODIC that put sensor in a unrecoverable state until power source restart.

I have checked both branchs with same results.

Dinidrol, 2020-05-19:
Hello! I have are same problem with SI7021.
But in i2c bus have ds3231. RTC read and setup the time. When i add si7021 
[image]

Dinidrol, 2020-05-19:
I tried connect of HMC5883L and DS3231 and work okey.
May by some problem on lib for SI7021?

legoater, 2021-01-06:
> SHT3x: Measurement is not started

@jmpmscorp , you could try this patch : https://github.com/legoater/esp-idf-lib/commit/88165987a872468d42e4a64311b1e2452996d51d. It fixes the `SHT3x: Measurement is not started` error on my little project.

I also see I2C init errors (device not being found) but only at board reset and I need to power cycle the board to fix that. I suppose the bus is busy and needs a reset.

UncleRus, 2021-01-07:
@legoater Thank you! Your patch has been merged.

closed 2021-03-26 by UncleRus