UncleRus/esp-idf-lib #86: HELP i2cdev

SERIDJ, 2020-06-04:
Hi,

I thank you very much for the work you did for those code es-idf-lib

So in my project i want use both hardware i2c port to use many device i2c  (8 device at the same time : 4 with port 0 and 4 with port 1 ) 

-I need to understand how it work i2cdev, i cant see why we use mutex. 

-And i dont understand too how u initialize with i2cdev_init() bcp i cant see those parm : 

/**************************************************/
		i2c_config_t conf;
		conf.mode = I2C_MODE_MASTER;
		conf.sda_io_num = SDA;
		conf.scl_io_num = SCL;
		conf.sda_pullup_en = GPIO_PULLUP_DISABLE; //GPIO_PULLUP_ENABLE;
		conf.scl_pullup_en = GPIO_PULLUP_DISABLE; //GPIO_PULLUP_ENABLE;
		conf.master.clk_speed = 100000;
		i2c_param_config(BUS-1, &conf);
		i2c_driver_install(BUS-1, I2C_MODE_MASTER, 0, 0, 0);
/**************************************************/

-If u can help i can add easily other device like mcp3428 

Best regards,

UncleRus, 2020-06-17:
Hi!

i2cdev uses two types of mutexes: port and transactional.

Port mutexes (there are only two of them, by the number of I2C ports) are necessary to avoid problems in the situation mentioned in the documentation:

> The I2C APIs are not thread-safe, if you want to use one I2C port in different tasks, you need to take care of the multi-thread issue.

They are taken before executing single I2C transactions and are released immediately after them.

Mutexes of the second type are created one per device and are necessary if the same device is used in several tasks:

```C
static i2c_dev_t some_dev;

void task1 (void * arg)
{
     // some_dev used here
}

void task2 (void * arg)
{
     // and here some_dev used too
}
```

In the vast majority of cases, this does not pose any problems (if there are port mutexes), but sometimes it is critically important: for example, if single device operation requires several I2C transactions in a row.

UncleRus, 2020-06-18:
With i2cdev, you can use almost any way to connect I2C devices.
For example, in the case of using SSD1306 and two MCP3428s, I would recommend connecting them like this:
- 2 GPIO outputs on ESP32 for dedicated screen connection via I2C bus 0
- 2 GPIO outputs to the second I2C bus, to which 2 MCP3428 are connected with different addresses.

If you need to connect more than one sensor with the same addresses and there are free GPIOs, then you can directly connect these sensors to individual GPIO outputs instead of using the I2C multiplexer. i2cdev will take care of reconfiguring I2C driver to the according outputs when you exchange data with these devices.

SERIDJ, 2020-06-18:
@UncleRus  thanks for ur replay, i tested a loot of device like bme280,mcp23017,bh175,... that work well  :) :) but i cant find SSD1306 with i2cdev ...

so for me i need to use mux bcp i want connect until 8 device with the same adresse ...
i write i simple code for selection : 

void SetI2CConnection(uint8_t i) {
  if (i > 7) return;
  i2c_cmd_handle_t cmd = i2c_cmd_link_create ( ) ;
  i2c_master_start ( cmd ) ;
  i2c_master_write_byte ( cmd ,  ( 0x70 <<  1 )  | I2C_MASTER_WRITE ,  true ) ;
  i2c_master_write_byte ( cmd , 1 << i ,  true );
  i2c_master_stop ( cmd ) ;
  esp_err_t res =i2c_master_cmd_begin ( 0 , cmd ,  1000  / portTICK_RATE_MS ) ;
      if (res != ESP_OK)
               ESP_LOGE(TAG, "***************************"
            		   	   	 "***************************"
            		   	   	 "***************************"
            		   	  "Could not read from mux TCA9548A device"
            		   	   	  "***************************"
            		          "***************************"
            		          "***************************"
            		   );
  i2c_cmd_link_delete(cmd);

}****

i call  SetI2CConnection(x)  in each task to select the channel device it work with bm280 and bh1750 but i have error with mcp23017 task : 

/********************************************work****************************************************//
void Lux_bh1750(void *pvParameters)
{
    i2c_dev_t dev;
    memset(&dev, 0, sizeof(i2c_dev_t)); // Zero descriptor

    bh1750_init_desc(&dev,0x23,0,21,22); //0x5c
    bh1750_setup(&dev, BH1750_MODE_CONTINIOUS, BH1750_RES_HIGH);

    while (1)
    {
		xSemaphoreTake( xMutex, portMAX_DELAY );
		SetI2CConnection(2);
		xSemaphoreGive( xMutex );

        uint16_t lux;

        if (bh1750_read(&dev, &lux) != ESP_OK)
            printf("Could not read lux data\n");
        else
            printf("Lux: %d\n", lux);

        vTaskDelay(3000 / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL);

}
/**********************************************error **************************************************//

void task_gpio_expander_mcp23017(void *pvParameters2)
{
     ESP_LOGI(TAG, "task task_gpio_expander_mcp23017\n");
     mcp23x17_t dev;
     memset(&dev, 0, sizeof(mcp23x17_t));
     mcp23x17_init_desc(&dev, 0, 0x20/*MCP23X17_ADDR_BASE*/, 21, 22);

     vTaskDelay(2000 / portTICK_PERIOD_MS);

     xSemaphoreTake( xMutex, portMAX_DELAY );
     SetI2CConnection(1);
	xSemaphoreGive( xMutex );

      vTaskDelete(NULL);

}
/***********************************************************************************************///

have this error like driver is not installed and i cant install it bcs i2cdev do that in each task   : 

E (16314) i2c: C:/Users/mohammed.seridj/esp/esp-idf/components/driver/i2c.c:1273 (i2c_master_cmd_begin):i2c driver not installed
E (16314) Main app: *********************************************************************************Could not read from mux TCA9548A device*********************************************************************************
E (16334) I2C_DEV: Could not read from device [0x20 at 0]: -1
E (17334) I2C_DEV: Could not read from device [0x20 at 0]: ****

so if u install driver inside task_device how i cant use another device using simple i2c bus init 

like : 

void i2c_init_bus(int BUS) {
	vTaskDelay(20 / portTICK_PERIOD_MS);
	switch (BUS) {
						case 1:
							   SDA =BUS1_SDA_PIN;
							   SCL =BUS1_SCL_PIN;
							   break;
						case 2:
							   SDA =BUS2_SDA_PIN;
							   SCL =BUS2_SCL_PIN;
							   break;
						}

		ESP_LOGD(tag, ">> i2c_init_scanner_bus");
		i2c_config_t conf;
		conf.mode = I2C_MODE_MASTER;
		conf.sda_io_num = SDA;
		conf.scl_io_num = SCL;
		conf.sda_pullup_en = GPIO_PULLUP_DISABLE; //GPIO_PULLUP_ENABLE;
		conf.scl_pullup_en = GPIO_PULLUP_DISABLE; //GPIO_PULLUP_ENABLE;
		conf.master.clk_speed = 100000;
		i2c_param_config(BUS-1, &conf);
		i2c_driver_install(BUS-1, I2C_MODE_MASTER, 0, 0, 0);
}

UncleRus, 2020-06-19:
By the way, I added a driver for TCA9548 to make it convenient.

SERIDJ, 2020-06-19:
@UncleRus yes yes i see it thaks  , if i want use 2 same device connected to ch0 and ch1 TCA9548 i use one task and i change inside the channels or i need t create  task for each bme ?

UncleRus, 2020-06-19:
For many devices switched via an I2C multiplexer, I would write something like this:

```C
...
#define SENSORS_CNT 2

static bme680_t sensors[SENSORS_CNT] = { 0 };
static i2c_dev_t i2cmux;
...

void task1(void *arg)
{
    // we can use i2c devices here:
    
    uint32_t duration;
    bme680_get_measurement_duration(&sensors[0], &duration);

    TickType_t last_wakeup = xTaskGetTickCount();
    bme680_values_float_t values;
    
    while (1)
    {
        for (size_t idx = 0; idx < SENSORS_CNT; idx++)
        {
            // switch mux
            tca9548_set_channels(1 << idx)
            // measure
            if (bme680_force_measurement(&sensors[0]) == ESP_OK)
            {
                vTaskDelay(duration);
                if (bme680_get_results_float(&sensors[0], &values) == ESP_OK)
                    printf("BME680 Sensor: %.2f °C, %.2f %%, %.2f hPa, %.2f Ohm\n",
                            values.temperature, values.humidity, values.pressure, values.gas_resistance);
            }
        }
        vTaskDelayUntil(&last_wakeup, 1000 / portTICK_PERIOD_MS);
    }
   
}

void task2(void *arg)
{
    // and here we can use i2c devices too
    ....
}

void app_main()
{
    ...
    ESP_ERROR_CHECK(i2cdev_init());
    
    memset(sensors, 0, sizeof(sensors));
    memset(&i2cmux, 0, sizeof(i2cmux));
    ESP_ERROR_CHECK(tca9548_init_desc(&i2cmux, PORT, MUX_ADDR, SDA_GPIO, SCL_GPIO));
    
    for (size_t i = 0; i < SENSORS_CNT; i++)
    {
        // switch i2c channel
        ESP_ERROR_CHECK(tca9548_set_channels(1 << i));
        
        // Init descriptor and device
        ESP_ERROR_CHECK(bme680_init_desc(&sensors[i], ADDR, PORT, SDA_GPIO, SCL_GPIO));
        ESP_ERROR_CHECK(bme680_init_sensor(&sensors[i]));
        ...
    }
    ...
    // create tasks
    ...
}

```

SERIDJ, 2020-06-20:
Ah yes it’s work thanks, how about OLED SSD1306 I can’t find library with ur driver i2cdev ?

UncleRus, 2020-06-23:
It's simple: use a third-party component for SSD1306, for example https://github.com/TaraHoleInIt/tarablessd1306. Configure it on a separate I2C bus, for example, 0. Use all other devices with i2cdev on bus 1.

SERIDJ, 2020-06-24:
I try with only Helloworld program and it’s not work well [image]

SERIDJ, 2020-06-25:
and i u can please convert my library mcp3428 to i2cdev : 

#define ACK_VAL 0x0
#define NACK_VAL 0x1

static const char *TAG = "MCP3428";

int adc_mcp3428_read(Channel_Selection_t chx,Sample_Rate_Selection_t resolutionx,Gain_Selection_t Gainx) {
	
	uint8_t Config_Register =0b00010000;//RDY: Ready Bit = 0 and O/C: Conversion Mode Bit = 1
	Config_Register = (Config_Register | (chx<<5) | (resolutionx<<2) | (Gainx) );

	//printf("Config_Register : HEX %0x and DEC %d\n",Config_Register,Config_Register);

    vTaskDelay(100 / portTICK_PERIOD_MS);
    /********************************************************************************************/
    i2c_cmd_handle_t cmd = i2c_cmd_link_create ( ) ;
    i2c_master_start ( cmd ) ;
    i2c_master_write_byte ( cmd ,  ( 0x69 <<  1 )  | I2C_MASTER_WRITE ,  true ) ;
    i2c_master_write_byte ( cmd , Config_Register ,  true );//0x10 ch1 //0x30 ch2 //0x50 ch3 //0x 70 ch4
   // i2c_master_write_byte ( cmd ,  0x00 ,  true );
    //i2c_master_write ( cmd , data_array , data_size ,  true ) ;

    //i2c_master_stop ( cmd ) ;
    //i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
     /********************************************************************************************/
    uint8_t first_byte , second_byte ;
    //cmd = i2c_cmd_link_create ( ) ;
    i2c_master_start ( cmd ) ;
    i2c_master_write_byte ( cmd ,  ( 0x69  <<  1 )  | I2C_MASTER_READ ,  true ) ;
    i2c_master_read_byte ( cmd ,  & first_byte , ACK_VAL ) ;
    i2c_master_read_byte ( cmd ,  & second_byte ,NACK_VAL ) ;
    i2c_master_stop ( cmd ) ;
    esp_err_t res =i2c_master_cmd_begin ( I2C_NUM_0 , cmd ,  1000  / portTICK_RATE_MS ) ;
    if (res != ESP_OK)
             ESP_LOGE(TAG, "Could not read from device");
    i2c_cmd_link_delete ( cmd ) ;
    /********************************************************************************************/
	//printf("first byte :  %d\n",first_byte);
	//printf("second byte : %d\n",second_byte);
	// Convert the data to 12-bits
	int raw_adc = (( first_byte & 0x0F) * 256 +  second_byte);
    /********************************************************************************************/
    //printf("raw_adc:%d\n",raw_adc);
	return raw_adc;

}

/*********************************************************************************************************************************/

#ifndef MCP3428_H_
#define MCP3428_H_

#define ACK_VAL 0x0
#define NACK_VAL 0x1

typedef enum {
	 CHANNEL1  =  0x00,      /*!<  (Default) */
	 CHANNEL2  =  0x01,      /*!< */
	 CHANNEL3  =  0x02,      /*!<  (MCP3428 only, treated as Ã¢â‚¬Å“00Ã¢â‚¬ï¿½ by the MCP3426/MCP3427)*/
	 CHANNEL4,               /*!<  (MCP3428 only, treated as Ã¢â‚¬Å“00Ã¢â‚¬ï¿½ by the MCP3426/MCP3427)*/
} Channel_Selection_t;

typedef enum {
    BITS_12 = 0x00,      /*!< 240 SPS (12 bits) (Default) */
	BITS_14 = 0x01,      /*!<  60 SPS (14 bits)*/
	BITS_16,			 /*!<   15 SPS (16 bits)*/

}  Sample_Rate_Selection_t;

typedef enum {
    GAIN_1 = 0x00,      /*!< x1 (Default) */
	GAIN_2 = 0x01,      /*!< x2*/
	GAIN_4 = 0x02,      /*!< x4*/
	GAIN_8,			    /*!< x8*/
} Gain_Selection_t;

typedef struct{
	Channel_Selection_t CH;                     /*!< */
	Sample_Rate_Selection_t resolution ;        /*!<  */
	Gain_Selection_t Gain;            			/*!< */
}mcp3428_config_t;

int adc_mcp3428_read(Channel_Selection_t chx,Sample_Rate_Selection_t resolutionx,Gain_Selection_t Gainx) ;

#endif /* MCP3428_H_ */
/** @}*/

/*********************************************************************************************************************************/

UncleRus, 2020-06-25:
> and i u can please convert my library mcp3428 to i2cdev

Please create a separate issue.

I also recommend that you read the article: https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

SERIDJ, 2020-06-26:
@UncleRus  oky thanks, its done

closed 2020-07-08 by UncleRus