UncleRus/esp-idf-lib #90: BME680 example gives zero values for temp, humidity and pressure with ESP32

bastianhjaeger, 2020-06-14:
**Device type**
- [x] ESP32 (DOIT dev board) --> with error
- [ ] ESP8226 (ESP12f) --> working

**Framework version**
- [x] ESP-IDF 3a9228fa02e933a28b15a335480fcc5e6f9d567a
- [x] esp-idf-lib 438dbbb21fe506dc895c972031f9e19a3d8fcf40

**Describe the bug**
I am seeing different results running the BME680 example code on an ESP12f and ESP32. Siee los attached. 

Both chips recognize the sensor via I2C, but the ESP32 keeps converting the RAW data to zero
```
D (1774) BME680: Started measurement
D (1924) BME680: Raw data: 505721 524288 20095 873 8
D (1924) BME680: Fixed point sensor values - 0/100 deg.C, 0/1000 %, 0 Pa, 24661 Ohm
BME680 Sensor: 0.00 °C, 0.00 %, 0.00 hPa, 24661.00 Ohm
```

The ESP12f on the other hand works fine:
```
D (2458) BME680: Started measurement
D (2599) BME680: Raw data: 502435 524288 20547 462 10
D (2601) BME680: Fixed point sensor values - 2704/100 deg.C, 50947/1000 %, 0 Pa, 8099 Ohm
BME680 Sensor: 27.04 °C, 50.95 %, 0.00 hPa, 8099.00 Ohm
```

The BME is powerd by 3.3V from the board. 
I an not sure if there could be something in the sdkconfig that messes up the configuration of the sensor / i2c, but I can not imagine what that would be.

[sdkconfig.defaults.esp12f.txt](https://github.com/UncleRus/esp-idf-lib/files/4776920/sdkconfig.defaults.esp12f.txt)
[sdkconfig.defaults.esp32.txt](https://github.com/UncleRus/esp-idf-lib/files/4776921/sdkconfig.defaults.esp32.txt)
[esp12f.log](https://github.com/UncleRus/esp-idf-lib/files/4776905/esp12f.log)
[esp32.log](https://github.com/UncleRus/esp-idf-lib/files/4776906/esp32.log)

UncleRus, 2020-06-17:
Hmmm... Could you print `dev->calib_data (bme680_calib_data_t)` contains?

bastianhjaeger, 2020-06-17:
Interesting idea. My code from the esp8266 and the esp32 only differ by the pins. 
[image]

Not sure it there need to be more. But you are right. 

```
BME680 Sensor: 0.00 °C, 0.00 %, 0.00 hPa, 14696.00 Ohm
sensor.calib_data.par_t1:  0
sensor.calib_data.par_t2:  0
sensor.calib_data.par_t3:  0
sensor.calib_data.par_p1:  0
sensor.calib_data.par_p2:  0
sensor.calib_data.par_p3:  0
sensor.calib_data.par_p4:  0
sensor.calib_data.par_p5:  0
sensor.calib_data.par_p7:  0
sensor.calib_data.par_p6:  0
sensor.calib_data.par_p8:  0
sensor.calib_data.par_p9:  0
sensor.calib_data.par_p10: 0
sensor.calib_data.par_h1:  0
sensor.calib_data.par_h2:  0
sensor.calib_data.par_h3:  0
sensor.calib_data.par_h4:  0
sensor.calib_data.par_h5:  0
sensor.calib_data.par_h6:  0
sensor.calib_data.par_h7:  0
sensor.calib_data.par_gh1: 0
sensor.calib_data.par_gh2: 0
sensor.calib_data.par_gh3: 0
sensor.calib_data.t_fine:  0
sensor.calib_data.res_heat_range: 1
sensor.calib_data.res_heat_val:   40
sensor.calib_data.range_sw_err:   1
```

So what can this mean?

UncleRus, 2020-06-18:
Calibration data seems to be read or interpreted incorrectly.
Could you dump contents of uint8_t buf[BME680_CDM_SIZE] (bme680_init_sensor()) after reading it at line 580 in bme680.c?

bastianhjaeger, 2020-06-18:
Sure. 

Code:
[image]

ESP32 result:
```
dump uint8_t buf[BME680_CDM_SIZE] 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 28 aa 16 4c 13 03 66 6c 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 40 170 22 76 19 3 102 108 

W (170) BME680: Heater is not stable
```
ESP8266:
```
dump uint8_t buf[BME680_CDM_SIZE] 
00 60 67 03 0f 94 8f 07 d7 58 ff c7 1b b7 ff 34 1e 00 00 01 f1 c3 f7 1e 01 40 3d 2a 00 2d 14 78 9c be 65 3d e7 ce 12 5b 00 29 aa 16 cb f3 fc 62 67 
0 96 103 3 15 148 143 7 215 88 255 199 27 183 255 52 30 0 0 1 241 195 247 30 1 64 61 42 0 45 20 120 156 190 101 61 231 206 18 91 0 41 170 22 203 243 252 98 103 

W (358) BME680: Heater is not stable

```

bastianhjaeger, 2020-06-18:
@UncleRus I found a "fix" with all your hints. 
Increasing `BME680_RESET_PERIOD` from 5 to 10 ms 
or 
adding an extra 10 ms delay before the read calls (`i2c_dev_read_reg(..)`) around line 582 of bme680.c helps.

closed 2020-06-19 by UncleRus

UncleRus, 2020-06-19:
Great! Thank you!

commit 3c8573ac (2020-06-19): bme680: fix #90