UncleRus/esp-idf-lib #91: DRIVER FOR mcp3428

SERIDJ, 2020-06-26:
I am requesting a driver for `mcp3428`.

### Device description

`MCP3428` is a chip to do a great thing. It supports only 2C. It
.

### Device information

The product information and the data sheet can be found at:

https://webge.github.io/MCP342x/

### Purchasing the product

The device can be purchased at:

-  Aliexpress
-  Amazon
-  ebay
-  Digi-Key

and i u can please convert my library mcp3428 to i2cdev : 

```
#define ACK_VAL 0x0
#define NACK_VAL 0x1

static const char *TAG = "MCP3428";

int adc_mcp3428_read(Channel_Selection_t chx,Sample_Rate_Selection_t resolutionx,Gain_Selection_t Gainx) {
	
	uint8_t Config_Register =0b00010000;//RDY: Ready Bit = 0 and O/C: Conversion Mode Bit = 1
	Config_Register = (Config_Register | (chx<<5) | (resolutionx<<2) | (Gainx) );

	//printf("Config_Register : HEX %0x and DEC %d\n",Config_Register,Config_Register);

    vTaskDelay(100 / portTICK_PERIOD_MS);
    /********************************************************************************************/
    i2c_cmd_handle_t cmd = i2c_cmd_link_create ( ) ;
    i2c_master_start ( cmd ) ;
    i2c_master_write_byte ( cmd ,  ( 0x69 <<  1 )  | I2C_MASTER_WRITE ,  true ) ;
    i2c_master_write_byte ( cmd , Config_Register ,  true );//0x10 ch1 //0x30 ch2 //0x50 ch3 //0x 70 ch4
   // i2c_master_write_byte ( cmd ,  0x00 ,  true );
    //i2c_master_write ( cmd , data_array , data_size ,  true ) ;

    //i2c_master_stop ( cmd ) ;
    //i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
     /********************************************************************************************/
    uint8_t first_byte , second_byte ;
    //cmd = i2c_cmd_link_create ( ) ;
    i2c_master_start ( cmd ) ;
    i2c_master_write_byte ( cmd ,  ( 0x69  <<  1 )  | I2C_MASTER_READ ,  true ) ;
    i2c_master_read_byte ( cmd ,  & first_byte , ACK_VAL ) ;
    i2c_master_read_byte ( cmd ,  & second_byte ,NACK_VAL ) ;
    i2c_master_stop ( cmd ) ;
    esp_err_t res =i2c_master_cmd_begin ( I2C_NUM_0 , cmd ,  1000  / portTICK_RATE_MS ) ;
    if (res != ESP_OK)
             ESP_LOGE(TAG, "Could not read from device");
    i2c_cmd_link_delete ( cmd ) ;
    /********************************************************************************************/
	//printf("first byte :  %d\n",first_byte);
	//printf("second byte : %d\n",second_byte);
	// Convert the data to 12-bits
	int raw_adc = (( first_byte & 0x0F) * 256 +  second_byte);
    /********************************************************************************************/
    //printf("raw_adc:%d\n",raw_adc);
	return raw_adc;

}
```

UncleRus, 2020-06-27:
See #92

SERIDJ, 2020-06-29:
@UncleRus yes its work thanks ,

closed 2020-06-30 by UncleRus