UncleRus/esp-idf-lib #93: Can't add the components - fatal error: hd44780.h: No such file or directory

vlaskz, 2020-07-06:
I'm trying to add the components as described in the readme.md:
```
CMAKELISTS:

set(COMPONENT_SRCS "hello_world_main.c")
set(COMPONENT_ADD_INCLUDEDIRS "")
set(EXTRA_COMPONENT_DIRS /~/esp/esp-idf-lib/components)
register_component()

MAKEFILE:

PROJECT_NAME := hello-world
EXTRA_COMPONENT_DIRS := /~/esp/esp-idf-lib/components
include $(IDF_PATH)/make/project.mk

```
[image]

and my components folder is in: ~/esp/esp-idf-lib/components:

[image]

but, when i do  `make flash` all i receive is: `fatal error: hd44780.h: No such file or directory`

UncleRus, 2020-07-06:
Wrong path in `EXTRA_COMPONENTS_DIR`.
Please use full path like this:
`EXTRA_COMPONENTS_DIR = /home/<username>/esp/esp-idf-lib/components`

closed 2020-07-06 by UncleRus

vlaskz, 2020-07-07:
thank you, Mr. You're awesome.

vlaskz, 2020-07-08:
Hello Mr. i would appreciate your help one more time.  I just tested the solution you've provided now, and it doesn't work. It still showing 'no such file or directory' error.