UncleRus/esp-idf-lib #94: ADS111x always busy

schroedermarius, 2020-07-06:
**Device type**
- [x] ESP32
- [] ESP8226

**Framework version**
- [x] ESP-IDF latest master
- [] ESP-IDF 3.2.x
- [] ESP-IDF 3.3.x
- [] ESP8266 RTOS SDK latest master
- [] ESP8266 RTOS SDK 3.2

**Describe the bug**
Im trying to connect the ADS1115 chip to my ESP32 board on ESP-IDF using your library.
First of all I tried your example code for the library which checks the ADS for readiness (ads111x_is_busy) which was always true. So the chip was busy.
I've figured out to call "ads111x_start_conversion" before "ads111x_is_busy" to achieve the chip to be ready. Should it be done like that?

Now I've got the problem, that the raw ADC value printed out by "ads111x_get_value" is always 1.
Im trying to measure pressure.

The code im setting up my device for the first pressure sensor is the following:
`
    ESP_ERROR_CHECK(ads111x_init_desc(device, ADS111X_ADDR_GND, I2C_NUM_0, GPIO_NUM_21, GPIO_NUM_22));
    ESP_ERROR_CHECK(ads111x_set_mode(device, ADS111X_MODE_SINGLE_SHOT));
    ESP_ERROR_CHECK(ads111x_set_gain(device, ADS111X_GAIN_6V144));
    ESP_ERROR_CHECK(ads111x_set_input_mux(pressure_up1, ADS111X_MUX_0_1));
`

schroedermarius, 2020-07-06:
I've changed it to `ADS111X_MUX_0_GND` and got now good measuring values around 2890.
I've only attached 1 sensor at ADS111X_MUX_0_GND, why am I getting the same values at ADS111X_MUX_1_GND, ADS111X_MUX_2_GND and ADS111X_MUX_3_GND? I need to switch the channels for other sensors.

helopez, 2020-09-01:
I had the same problem (reading the same voltage) when trying to measure a differential and a relative to ground voltage. Is there a work around to this problem?

helopez, 2020-09-02:
I figured out the problem, this is not a bug, at least not a problem with the library.

Depending on how you set the ads111x_set_data_rate, you must add a delay to allow the ads1115 chip to measure the value. In my case I was setting the data rate to 8 per second (ADS111X_DATA_RATE_8) , theoretically then I must wait 125ms before reading the value. I configured it to 150ms and the values are read fine.
Another way of doing this would be to connect the alert/ready pin to get notified when the conversion is ready.

DhruvaG2000, 2021-03-05:
Hey feel free to refer [this repo](https://github.com/DhruvaG2000/ESP32-ADS1115) where you will find pin diagram and corresponding code that actually works!

closed 2021-03-05 by UncleRus

HamadAli09, 2024-03-10:
Hi, I see this issue is closed but i am stuck at this point. I am only able to read one  analaog input channel at one time only. How can i read all four single ended channels simultaneously?