UncleRus/esp-idf-lib #98: multi-sensor

SERIDJ, 2020-07-20:
hi,

i have i bus i2c in my board and i want to detect sensor connected for each socket and start it but i dont know how i can write this code, now i have table with device found and from this table i want start read data sensor connected ( i can connect from 1 until 8 device  same or different type ) 

[image]

UncleRus, 2020-07-23:
Hi! You need separate descriptors for each i2c device.

SERIDJ, 2020-07-23:
@UncleRus  oky so i work with detected device so i can create instance with malloc bcs number of device is detected with i2ccannser 

now i create max : 

/***************************************************/
static float pressure, temperature, humidity;
static bmp280_t            dev_bme280 [8]  =  {0};
static bmp280_params_t params_bme280  [8]   = {0};
/****************************************************/
uint16_t lux;
i2c_dev_t dev_bh1750[8]=  {0};
/****************************************************/
i2c_dev_t dev_pcf8574[8];
uint16_t read_port_pcf8574;

/****************************************************/

UncleRus, 2020-07-23:
Yes you can create device descriptors dynamically, e.g.
```C
i2c_dev_t *bh1750; // pointer to device descriptor

...

bh1750 = malloc(sizeof(i2c_dev_t)); // allocate memory for descriptor
memset(bh1750, 0, sizeof(i2c_dev_t))); // fill it with zeroes

... // init descriptor, init device, use it

free(bh1750); // free allocated memory
```

SERIDJ, 2020-07-23:
Ah yes yes but for sensor i don't think I free memory bcs I use all time read_bh1750 and I don't initialize device for each read 

So it's don't give problem if I don't use free(p) ?

closed 2021-03-12 by UncleRus