UncleRus/esp-idf-lib #109: I2c could not read from device error for mcp23107

AjayArumugam07, 2020-08-23:
**Device type**
- [x] ESP32

**Framework version**
- [x] ESP-IDF 4.0

**Describe the bug**
I am trying to expand the existing gpio ports on my esp32 using the mcp23107. I have connected LEDs to gpio ports 6 and 2 of the MCP23107 and they are blinking as expected. 
However, I get this error once everytime I run the program: E (1575) I2C_DEV: Could not read from device [0x20 at 0]: 263

I have connected A1, A2, A3 to ground so the address should be 0x20. 

Here is the code I used:
ESP_ERROR_CHECK(i2cdev_init());
		    mcp23x17_t dev;
		    memset(&dev, 0, sizeof(mcp23x17_t));

		    ESP_ERROR_CHECK(mcp23x17_init_desc(&dev, 0, MCP23X17_ADDR_BASE, RTC_SDA_GPIO, RTC_SCL_GPIO));

		    // Setup PORTA0 as input
		    mcp23x17_set_mode(&dev, 0, MCP23X17_GPIO_INPUT);
		    // Setup interrupt on it
		    mcp23x17_set_interrupt(&dev, 0, MCP23X17_INT_ANY_EDGE);

		    gpio_set_direction(INTA_GPIO, GPIO_MODE_INPUT);
		    gpio_set_intr_type(INTA_GPIO, GPIO_INTR_ANYEDGE);
		    gpio_install_isr_service(0);
		    gpio_isr_handler_add(INTA_GPIO, intr_handler, NULL);

		    // Setup PORTB0 as output
		    mcp23x17_set_mode(&dev, 6, MCP23X17_GPIO_OUTPUT);
		    mcp23x17_set_mode(&dev, 2, MCP23X17_GPIO_OUTPUT);
		    bool on = true;
		    while (1)
		    {
		        mcp23x17_set_level(&dev, 6, on);
		        mcp23x17_set_level(&dev, 2, on);
		        on = !on;
		        vTaskDelay(500 / portTICK_PERIOD_MS);
		    }

ChristiaanHaas, 2020-11-12:
Did  you manage to resolve this? I'm having the same issue right now

UncleRus, 2020-11-12:
Could you please provide console output?

ChristiaanHaas, 2020-11-12:
Sure! This is my console output:
`E (5495) I2C_DEV: Could not read from device [0x20 at 0]: 263`

and this is the code I am using:
MCP_Chip.h
```cpp
class MCP_Chip {
public:
  explicit MCP_Chip();

  _Noreturn static void toggleEverySecond(void *arg);

private:
  TaskHandle_t taskHandle;
  mcp23x17_t dev;
};
```

MCP_Chip.cpp
```cpp
#include <cstring>
#include <Constants.h>
#include "MCP_Chip.h"
#include "esp_log.h"

#define TAG "MCP_Chip"

_Noreturn void MCP_Chip::toggleEverySecond(void *arg) {
  auto chip = (MCP_Chip*) arg;
  bool on = true;
  while (true) {
    mcp23x17_set_level(&chip->dev, 7, on);
    on = !on;
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
};

MCP_Chip::MCP_Chip() {
  ESP_LOGI(TAG, "Constructing MCP_Chip");
  ESP_ERROR_CHECK(mcp23x17_init_desc(&dev, 0, MCP23X17_ADDR_BASE, GPIO_NUM_21, GPIO_NUM_22));

  mcp23x17_set_mode(&dev, 7, MCP23X17_GPIO_OUTPUT);

  xTaskCreate(
      MCP_Chip::toggleEverySecond,
      "toggleEverySecond",
      8192 / sizeof(StackType_t),
      this,
      TASK_PRIORITY_HIGH,
      &this->taskHandle
  );
}
```

I assume the connections are set up properly, I connected it the same as @AjayArumugam07, moreover:
When I use the following function my mcp chip does seem to work:
```c
static esp_err_t writeMCP(i2c_port_t i2c_num, uint8_t i2c_reg, uint8_t *data_wr, size_t size) {
    int ret;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (MCP_SLAVE_ADDRESS << 1) | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, i2c_reg, ACK_CHECK_EN);
    i2c_master_write(cmd, data_wr, size, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    return ret;
}
```

ChristiaanHaas, 2020-11-12:
In the [mcp23017 example](https://github.com/UncleRus/esp-idf-lib/blob/master/examples/mcp23017/main/main.c) I did notice I am missing the following line:
```c
 memset(&dev, 0, sizeof(mcp23x17_t));
```
But adding that made no difference, I still get the same 263 error code. [Error code 263 = 0x107 is a timeout issue](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/error-codes.html) so I tried setting the I2C timeout to 5 seconds to see if that would fix it, but that also did not work.

UncleRus, 2020-11-14:
I have already seen reports that some ICs or boards are unstable at the default frequency.
Try lowering I2C clock frequency (default is 1MHz).
https://github.com/UncleRus/esp-idf-lib#how-can-i-change-frequency-of-i2c-clock-at-default-frequency-my-device-is-unstable-or-not-working-at-all

ChristiaanHaas, 2020-11-17:
Nice, that was indeed part of the problem, 10kHz is more stable than 1Mhz. But there are lots of other things that I had to change before I got the MCP to work. This is what I ended up with eventually:
mcp23x17.h
```cpp
/**
 * @file mcp23x17.h
 * @defgroup mcp23x17 mcp23x17
 * @{
 *
 * ESP-IDF driver for I2C/SPI 16 bit GPIO expanders MCP23017/MCP23S17
 *
 * Copyright (C) 2018 Ruslan V. Uss <https://github.com/UncleRus>
 *
 * BSD Licensed as described in the file LICENSE
 */
#pragma once

#include <stddef.h>
#include <stdbool.h>
#include <i2cdev.h>
#include <driver/spi_master.h>
#include <driver/gpio.h>
#include <esp_err.h>

#define MCP23X17_ADDR_BASE 0x20

#ifdef __cplusplus
extern "C" {
#endif

typedef i2c_dev_t mcp23x17_t;

/**
 * GPIO mode
 */
typedef enum {
  MCP23X17_GPIO_OUTPUT = 0,
  MCP23X17_GPIO_INPUT
} mcp23x17_gpio_mode_t;

/**
 * @brief Initialize device descriptior
 * SCL frequency is 1MHz
 * @param dev Pointer to device descriptor
 * @param port I2C port number
 * @param addr I2C address
 * @param sda_gpio SDA GPIO
 * @param scl_gpio SCL GPIO
 * @return `ESP_OK` on success
 */
esp_err_t mcp23x17_init(mcp23x17_t *dev, i2c_port_t port, uint8_t addr, gpio_num_t sda_gpio, gpio_num_t scl_gpio);

/**
 * @brief Free device descriptor
 * @param dev Pointer to device descriptor
 * @return `ESP_OK` on success
 */
esp_err_t mcp23x17_free_desc(mcp23x17_t *dev);

/**
 * @brief Read GPIO pin level
 * @param dev Pointer to device descriptor
 * @param pin Pin number, 0 for PORTA/GPIO0..15 for PORTB/GPIO7
 * @param[out] val `true` if pin currently in high state
 * @return `ESP_OK` on success
 */
esp_err_t mcp23x17_get_pin(mcp23x17_t *dev, uint8_t pin, bool *val);

/**
 * @brief Set GPIO pin level
 *
 * Pin must be set up as output
 *
 * @param dev Pointer to device descriptor
 * @param pin Pin number, 0 for PORTA/GPIO0..15 for PORTB/GPIO7
 * @param[out] val `true` if pin currently in high state
 * @return `ESP_OK` on success
 */
esp_err_t mcp23x17_set_pin(mcp23x17_t *dev, uint8_t pin, bool val);

esp_err_t mcp_set_all_output(mcp23x17_t *dev);

esp_err_t mcp_set_all_pins(mcp23x17_t *dev, bool enabled);

#ifdef __cplusplus
}
#endif
```
mcp23x17.c
```cpp
/**
 * @file mcp23x17.c
 *
 * ESP-IDF driver for I2C/SPI 16 bit GPIO expanders MCP23017/MCP23S17
 *
 * Copyright (C) 2018 Ruslan V. Uss <https://github.com/UncleRus>
 *
 * BSD Licensed as described in the file LICENSE
 */

#include <esp_log.h>
#include <esp_idf_lib_helpers.h>
#include "mcp23x17.h"

static const char *TAG = "MCP23x17";

#define I2C_FREQ_HZ 100000  // Max 1MHz for esp-idf, but device supports up to 1.7Mhz

#define REG_IODIRA   0x00
#define REG_IODIRB   0x01
#define REG_IPOLA    0x02
#define REG_IPOLB    0x03
#define REG_GPINTENA 0x04
#define REG_GPINTENB 0x05
#define REG_DEFVALA  0x06
#define REG_DEFVALB  0x07
#define REG_INTCONA  0x08
#define REG_INTCONB  0x09
#define REG_IOCON    0x0A
#define REG_GPPUA    0x0C
#define REG_GPPUB    0x0D
#define REG_INTFA    0x0E
#define REG_INTFB    0x0F
#define REG_INTCAPA  0x10
#define REG_INTCAPB  0x11
#define REG_GPIOA    0x12
#define REG_GPIOB    0x13
#define REG_OLATA    0x14
#define REG_OLATB    0x15

#define BIT_IOCON_INTPOL 1
#define BIT_IOCON_ODR    2
#define BIT_IOCON_HAEN   3
#define BIT_IOCON_DISSLW 4
#define BIT_IOCON_SEQOP  5
#define BIT_IOCON_MIRROR 6
#define BIT_IOCON_BANK   7

#define CHECK(x) do { esp_err_t __; if ((__ = x) != ESP_OK) return __; } while (0)
#define CHECK_ARG(VAL) do { if (!(VAL)) return ESP_ERR_INVALID_ARG; } while (0)
#define BV(x) (1 << (x))

esp_err_t mcp_set_all_output(mcp23x17_t *dev) {
  CHECK_ARG(dev);

  I2C_DEV_TAKE_MUTEX(dev);
  uint8_t data = 0x00;
  I2C_DEV_CHECK(dev, i2c_dev_write_reg(dev, REG_IODIRA, &data, 1));
  I2C_DEV_CHECK(dev, i2c_dev_write_reg(dev, REG_IODIRB, &data, 1));
  I2C_DEV_GIVE_MUTEX(dev);

  return ESP_OK;
}

esp_err_t mcp_set_all_pins(mcp23x17_t *dev, bool enabled) {
  CHECK_ARG(dev);

  I2C_DEV_TAKE_MUTEX(dev);
  uint16_t data = enabled ? 0xff : 0x00;
  I2C_DEV_CHECK(dev, i2c_dev_write_reg(dev, REG_IOCON, &data, 2));
  I2C_DEV_GIVE_MUTEX(dev);

  return ESP_OK;
}

static esp_err_t read_reg_bit_8(mcp23x17_t *dev, uint8_t reg, bool *val, uint8_t bit) {
  CHECK_ARG(dev && val);

  uint8_t buf;

  I2C_DEV_TAKE_MUTEX(dev);
  I2C_DEV_CHECK(dev, i2c_dev_read_reg(dev, reg, &buf, 1));
  I2C_DEV_GIVE_MUTEX(dev);

  *val = (buf & BV(bit)) >> bit;

  return ESP_OK;
}

static esp_err_t write_reg_bit_8(mcp23x17_t *dev, uint8_t reg, bool val, uint8_t bit) {
  CHECK_ARG(dev);

  uint8_t buf;

  I2C_DEV_TAKE_MUTEX(dev);
  I2C_DEV_CHECK(dev, i2c_dev_read_reg(dev, reg, &buf, 1));
  buf = (buf & ~BV(bit)) | (val ? BV(bit) : 0);
  I2C_DEV_CHECK(dev, i2c_dev_write_reg(dev, reg, &buf, 1));
  I2C_DEV_GIVE_MUTEX(dev);

  return ESP_OK;
}

esp_err_t mcp23x17_init(mcp23x17_t *dev, i2c_port_t port, uint8_t addr, gpio_num_t sda_gpio, gpio_num_t scl_gpio) {
  CHECK_ARG(dev);
  if (addr < MCP23X17_ADDR_BASE || addr > MCP23X17_ADDR_BASE + 7) {
    ESP_LOGE(TAG, "Invalid device address: 0x%02x", addr);
    return ESP_ERR_INVALID_ARG;
  }

  dev->port = port;
  dev->addr = addr;
  dev->cfg.mode = I2C_MODE_MASTER;
  dev->cfg.sda_io_num = sda_gpio;
  dev->cfg.sda_pullup_en = GPIO_PULLUP_ENABLE;
  dev->cfg.scl_io_num = scl_gpio;
  dev->cfg.scl_pullup_en = GPIO_PULLUP_ENABLE;
#if HELPER_TARGET_IS_ESP32
  dev->cfg.master.clk_speed = I2C_FREQ_HZ;
#endif

  return i2c_dev_create_mutex(dev);
}

esp_err_t mcp23x17_free_desc(mcp23x17_t *dev) {
  CHECK_ARG(dev);

  return i2c_dev_delete_mutex(dev);
}

esp_err_t mcp23x17_get_pin(mcp23x17_t *dev, uint8_t pin, bool *val) {
  CHECK_ARG(val);

  bool buf;
  CHECK(read_reg_bit_8(dev, REG_GPIOA, &buf, pin));
  *val = buf ? true : false;

  return ESP_OK;
}

esp_err_t mcp23x17_set_pin(mcp23x17_t *dev, uint8_t pin, bool val) {
  return write_reg_bit_8(dev, REG_IOCON, val, pin);
}

```

Bytesforus, 2020-12-24:
Just out of curiosity, I scanned the code quickly but I did not see the enabling of the ESP32 its internal pull-up resistors?

Especially at higher speeds I think these start to matter.

closed 2021-03-12 by UncleRus

Thecapn32, 2022-07-16:
I am having the exact same issue, I changed the frequency but It don't work

> Sure! This is my console output: `E (5495) I2C_DEV: Could not read from device [0x20 at 0]: 263`
> 
> and this is the code I am using: MCP_Chip.h
> 
> ```c++
> class MCP_Chip {
> public:
>   explicit MCP_Chip();
> 
>   _Noreturn static void toggleEverySecond(void *arg);
> 
> private:
>   TaskHandle_t taskHandle;
>   mcp23x17_t dev;
> };
> ```
> 
> MCP_Chip.cpp
> 
> ```c++
> #include <cstring>
> #include <Constants.h>
> #include "MCP_Chip.h"
> #include "esp_log.h"
> 
> #define TAG "MCP_Chip"
> 
> _Noreturn void MCP_Chip::toggleEverySecond(void *arg) {
>   auto chip = (MCP_Chip*) arg;
>   bool on = true;
>   while (true) {
>     mcp23x17_set_level(&chip->dev, 7, on);
>     on = !on;
>     vTaskDelay(1000 / portTICK_PERIOD_MS);
>   }
> };
> 
> MCP_Chip::MCP_Chip() {
>   ESP_LOGI(TAG, "Constructing MCP_Chip");
>   ESP_ERROR_CHECK(mcp23x17_init_desc(&dev, 0, MCP23X17_ADDR_BASE, GPIO_NUM_21, GPIO_NUM_22));
> 
>   mcp23x17_set_mode(&dev, 7, MCP23X17_GPIO_OUTPUT);
> 
>   xTaskCreate(
>       MCP_Chip::toggleEverySecond,
>       "toggleEverySecond",
>       8192 / sizeof(StackType_t),
>       this,
>       TASK_PRIORITY_HIGH,
>       &this->taskHandle
>   );
> }
> ```
> 
> I assume the connections are set up properly, I connected it the same as @AjayArumugam07, moreover: When I use the following function my mcp chip does seem to work:
> 
> ```c
> static esp_err_t writeMCP(i2c_port_t i2c_num, uint8_t i2c_reg, uint8_t *data_wr, size_t size) {
>     int ret;
>     i2c_cmd_handle_t cmd = i2c_cmd_link_create();
>     i2c_master_start(cmd);
>     i2c_master_write_byte(cmd, (MCP_SLAVE_ADDRESS << 1) | WRITE_BIT, ACK_CHECK_EN);
>     i2c_master_write_byte(cmd, i2c_reg, ACK_CHECK_EN);
>     i2c_master_write(cmd, data_wr, size, ACK_CHECK_EN);
>     i2c_master_stop(cmd);
>     ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
>     i2c_cmd_link_delete(cmd);
>     return ret;
> }
> ```