UncleRus/esp-idf-lib #118: SHT3x measurements fail after 72min

david-collett, 2020-09-09:
**Device type**
- [x] ESP8226

**Framework version**
- [x] ESP8266 RTOS SDK latest master

**Describe the bug**
Once the board has been running for 72min, trying to measure temp with `sht3x_measure` will fail with:

```E (4324210) SHT3x: Measurement is still running```

I believe it would fail immediately if the time was actually set. This is due to an integer overflow in the method  `get_time_us()` which returns a uint32_t which is not big enough to hold usec of more than 72min past zero.

I could attempt a patch, but I noticed the contributing author seems to have a newer version which may not exhibit the issue so it might be better to sync with that?

UncleRus, 2020-09-09:
Could you try a possible fix in #119 ?

david-collett, 2020-09-09:
That seems to have fixed it, thanks!

commit 75d5aacd (2020-09-09): Fixes #118
commit 7ecc160a (2020-09-09): Merge pull request #119 from UncleRus/fix_timer

Fixes #118
closed 2020-09-09 by UncleRus