UncleRus/esp-idf-lib #133: How can i add libraries in my projects?

aledoors, 2020-11-18:
Goodnight,
I am using an example program and I want to edit the code by adding some libraries, the code is as follows:

#include <stdio.h>
#include <stdbool.h>
#include <string.h>
#include "ssd1306.h"
#include "ssd1306_draw.h"
#include "ssd1306_font.h"
#include "ssd1306_default_if.h"
#include <esp_system.h>
#include <bmp280.h>
#include <string.h>

#define USE_I2C_DISPLAY
#define SDA_GPIO 4
#define SCL_GPIO 5

static const int I2CDisplayAddress = 0x3C;
static const int I2CDisplayWidth = 128;
static const int I2CDisplayHeight = 64;
static const int I2CResetPin = -1;

struct SSD1306_Device I2CDisplay;
void SetupDemo( struct SSD1306_Device* DisplayHandle, const struct SSD1306_FontDef* Font );
void SayHello( struct SSD1306_Device* DisplayHandle, const char* HelloText );

bool DefaultBusInit( void ) {
assert( SSD1306_I2CMasterInitDefault( ) == true );
assert( SSD1306_I2CMasterAttachDisplayDefault( &I2CDisplay, I2CDisplayWidth, I2CDisplayHeight, I2CDisplayAddress, I2CResetPin ) == true );

return true;
}
void SetupDemo( struct SSD1306_Device* DisplayHandle, const struct SSD1306_FontDef* Font ) {
SSD1306_Clear( DisplayHandle, SSD_COLOR_BLACK );
SSD1306_SetFont( DisplayHandle, Font );
}

void SayHello( struct SSD1306_Device* DisplayHandle, const char* HelloText ) {
SSD1306_FontDrawAnchoredString( DisplayHandle, TextAnchor_Center, HelloText, SSD_COLOR_WHITE );
SSD1306_Update( DisplayHandle );
}

void app_main( void ) {
printf( "Ready...\n" );

if ( DefaultBusInit( ) == true ) {
    printf( "BUS Init lookin good...\n" );
    printf( "Drawing.\n" );

    #if defined USE_I2C_DISPLAY
        SetupDemo( &I2CDisplay, &Font_droid_sans_fallback_24x28 );
        SayHello( &I2CDisplay, "SICCUS!" );
    #endif

    printf( "Done!\n" );
}
}

I'm using Eclipse IDE

but i get the following error when did the compilation:
Building in: D:\Eclipse\Trabajo\Oled_BMP280\build
cmake --build . -- -v
[1/9] cmd.exe /C "cd /D D:\Eclipse\Trabajo\Oled_BMP280\build\esp-idf\partition_table && "C:\Users\C. Alejandro.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo "Partition table binary generated. Contents:" && "C:\Users\C. Alejandro.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo ******************************************************************************* && python C:/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 4MB D:/Eclipse/Trabajo/Oled_BMP280/build/partition_table/partition-table.bin && "C:\Users\C. Alejandro.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo *******************************************************************************"
Partition table binary generated. Contents:

Espressif ESP32 Partition Table
Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
[... 7 lines cut ...]
ccache C:\Users\C765B1.ALE\ESPRES1\tools\XTENSA1\ESP-201.0\XTENSA1\bin\XT89941.EXE -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -IC:/esp-idf/components/newlib/platform_include -IC:/esp-idf/components/freertos/include -IC:/esp-idf/components/heap/include -IC:/esp-idf/components/log/include -IC:/esp-idf/components/soc/esp32/include -IC:/esp-idf/components/soc/include -IC:/esp-idf/components/esp_rom/include -IC:/esp-idf/components/esp_common/include -IC:/esp-idf/components/xtensa/include -IC:/esp-idf/components/xtensa/esp32/include -IC:/esp-idf/components/esp32/include -IC:/esp-idf/components/driver/include -IC:/esp-idf/components/esp_ringbuf/include -IC:/esp-idf/components/esp_event/include -IC:/esp-idf/components/tcpip_adapter/include -IC:/esp-idf/components/lwip/include/apps -IC:/esp-idf/components/lwip/include/apps/sntp -IC:/esp-idf/components/lwip/lwip/src/include -IC:/esp-idf/components/lwip/port/esp32/include -IC:/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp-idf/components/vfs/include -IC:/esp-idf/components/esp_wifi/include -IC:/esp-idf/components/esp_wifi/esp32/include -IC:/esp-idf/components/esp_eth/include -IC:/esp-idf/components/efuse/include -IC:/esp-idf/components/efuse/esp32/include -IC:/esp-idf/components/app_trace/include -IC:/esp-idf/components/mbedtls/port/include -IC:/esp-idf/components/mbedtls/mbedtls/include -IC:/esp-idf/components/wpa_supplicant/include -IC:/esp-idf/components/wpa_supplicant/port/include -IC:/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/esp-idf/components/bootloader_support/include -IC:/esp-idf/components/app_update/include -IC:/esp-idf/components/spi_flash/include -IC:/esp-idf/components/nvs_flash/include -IC:/esp-idf/components/pthread/include -IC:/esp-idf/components/espcoredump/include -IC:/esp-idf/components/asio/asio/asio/include -IC:/esp-idf/components/asio/port/include -IC:/esp-idf/components/coap/port/include -IC:/esp-idf/components/coap/port/include/coap -IC:/esp-idf/components/coap/libcoap/include -IC:/esp-idf/components/coap/libcoap/include/coap2 -IC:/esp-idf/components/console -IC:/esp-idf/components/nghttp/port/include -IC:/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp-idf/components/esp-tls -IC:/esp-idf/components/esp_adc_cal/include -IC:/esp-idf/components/esp_gdbstub/include -IC:/esp-idf/components/tcp_transport/include -IC:/esp-idf/components/esp_http_client/include -IC:/esp-idf/components/esp_http_server/include -IC:/esp-idf/components/esp_https_ota/include -IC:/esp-idf/components/protobuf-c/protobuf-c -IC:/esp-idf/components/protocomm/include/common -IC:/esp-idf/components/protocomm/include/security -IC:/esp-idf/components/protocomm/include/transports -IC:/esp-idf/components/mdns/include -IC:/esp-idf/components/esp_local_ctrl/include -IC:/esp-idf/components/esp_websocket_client/include -IC:/esp-idf/components/expat/expat/expat/lib -IC:/esp-idf/components/expat/port/include -IC:/esp-idf/components/wear_levelling/include -IC:/esp-idf/components/sdmmc/include -IC:/esp-idf/components/fatfs/diskio -IC:/esp-idf/components/fatfs/vfs -IC:/esp-idf/components/fatfs/src -IC:/esp-idf/components/freemodbus/common/include -IC:/esp-idf/components/idf_test/include -IC:/esp-idf/components/jsmn/include -IC:/esp-idf/components/json/cJSON -IC:/esp-idf/components/libsodium/libsodium/src/libsodium/include -IC:/esp-idf/components/libsodium/port_include -IC:/esp-idf/components/mqtt/esp-mqtt/include -IC:/esp-idf/components/openssl/include -IC:/esp-idf/components/spiffs/include -IC:/esp-idf/components/ulp/include -IC:/esp-idf/components/unity/include -IC:/esp-idf/components/unity/unity/src -IC:/esp-idf/components/wifi_provisioning/include -I../components/tarablessd1306 -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER="v4.0.1-dirty" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -MF esp-idf\main\CMakeFiles__idf_main.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -c ../main/main.c
In file included from ../components/tarablessd1306/ssd1306_err.h:5,
from ../components/tarablessd1306/ssd1306.h:11,
from ../main/main.c:11:
C:/esp-idf/components/xtensa/include/esp_panic.h:2:2: warning: #warning "esp_panic.h is deprecated, please use esp_debug_helpers.h or/and esp_private/panic_reason.h" [-Wcpp]
#warning "esp_panic.h is deprecated, please use esp_debug_helpers.h or/and esp_private/panic_reason.h"
^~~~~~~
In file included from C:/esp-idf/components/esp_common/include/esp_system.h:22,
from ../main/main.c:15:
C:/esp-idf/components/esp_common/include/esp_bit_defs.h:53: warning: "BIT" redefined
#define BIT(nr) (1UL << (nr))

In file included from ../main/main.c:11:
../components/tarablessd1306/ssd1306.h:19: note: this is the location of the previous definition
#define BIT( n ) ( 1 << n )

../main/main.c:16:10: fatal error: bmp280.h: No such file or directory
#include <bmp280.h>
^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Build complete (1 errors, 2 warnings): D:\Eclipse\Trabajo\Oled_BMP280\build

I think i have to do is add the library into my code, but i don't know to do it.
Thanks in advance

mrSilkie, 2020-11-18:
I am also stuck but am working on VS Code instead.

I have tried using the following with no success.
> #include "~/esp/esp-idf-lib/components/bmp280.h"
> #include "~/esp/esp-idf-lib/components/bmp280.c"

However, once I've copied and pasted the files into my main folder the following does work.
> #include "bmp280.c"

But i get the compile error
> ../main/bmp280.c:15:10: fatal error: esp_idf_lib_helpers.h: No such file or directory
 #include <esp_idf_lib_helpers.h>

I've yet to find a tutorial on this. When using eclipse the youtube 'esp technical tutorials' series on is really good and probably has the answers your looking for. https://www.youtube.com/watch?v=-ttiPfmrehU

However, i know that the question we're asking is a very basic c / makefile question and I'm sad to see somebody else struggling with the easy stuff since there are no tutorials for this on the internet.

trombik, 2020-11-18:
welcome to C, say goodbye to arduino.

https://www.tutorialspoint.com/cprogramming/

https://en.m.wikibooks.org/wiki/C_Programming

for your particular issue,
read:

https://www.tutorialspoint.com/cprogramming/c_header_files.htm

by the way, learning a computer language  by reading blogs and tutorials on the internet is not necessarily the best way.

The C programming language by K and R
https://en.m.wikipedia.org/wiki/The_C_Programming_Language

mrSilkie, 2020-11-18:
I've managed to get things to compile by changing my CMakeLists.txt to include the following and i think this is the correct way to do things

idf_component_register(SRCS "example.c"
                    INCLUDE_DIRS "~/esp/esp-idf-lib/components/bmp280"
                    INCLUDE_DIRS "~/esp/esp-idf-lib/components/esp_idf_lib_helpers"
                    INCLUDE_DIRS "~/esp/esp-idf-lib/components/i2cdev")

I am still getting undefined reference errors even after adding the include directories. We seem to be using the same libraries so if you ever get it working, please let me know.

> /home/user/esp/myprojects/Smart_Bucket/build/../main/smartbucket.c:99: undefined reference to `i2cdev_init'

aledoors, 2020-11-18:
> I've managed to get things to compile by changing my CMakeLists.txt to include the following and i think this is the correct way to do things
>
> idf_component_register(SRCS "example.c"
> INCLUDE_DIRS [...]

Unfortunately the little information that can still be found on the internet is not good and it is difficult to solve simple problems like the ones we find ourselves and we must fight a little to make it work. If I manage to get this problem out of the way, I will definitely comment here

aledoors, 2020-11-18:
> welcome to C, say goodbye to arduino.
>
> https://www.tutorialspoint.com/cprogramming/
>
> https://en.m.wikibooks.org/wiki/C_Programming
>
> for your particular issue,
> read: [...]

I am using freeRTOS which is a slightly different version of C and I think even the compiler works differently.

UncleRus, 2020-12-05:
Feel free to reopen it.

closed 2020-12-05 by UncleRus