UncleRus/esp-idf-lib #139: pcf8574 in platformio

RonKass, 2020-12-11:
Hi there.
I am trying to get my setup to work with the ESP-IDF and the esp-idf-lib components, specifically the PCF8574 io expander.
I am using a 'NodeMCU ESP32s' microcontroller.

The code I am using is as follows
`

    #include <stdio.h>
    #include <freertos/FreeRTOS.h>
    #include <freertos/task.h>
    #include <pcf8574.h>
    #include <string.h>

    #if defined(CONFIG_IDF_TARGET_ESP8266)
    #define SDA_GPIO 4
    #define SCL_GPIO 5
    #else
    #define SDA_GPIO 21
    #define SCL_GPIO 22
    #endif

    #define I2C_ADDR 32

    void test(void *pvParameters)
    {
        i2c_dev_t pcf8574;

        // Zero device descriptor
        memset(&pcf8574, 0, sizeof(i2c_dev_t));

        // Init i2c device descriptor
        ESP_ERROR_CHECK(pcf8574_init_desc(&pcf8574, 0, I2C_ADDR, SDA_GPIO, SCL_GPIO));

        // Do some blinking
        uint8_t port_val = 0xaa;
        while (1)
        {
            // invert value
            port_val = ~port_val;

            // write value to port
            pcf8574_port_write(&pcf8574, port_val);

            vTaskDelay(500 / portTICK_PERIOD_MS);
        }
    }

    void app_main()
    {
        // Init i2cdev library
        ESP_ERROR_CHECK(i2cdev_init());

        xTaskCreate(test, "test", configMINIMAL_STACK_SIZE * 5, NULL, 5, NULL);
    }
`

I've cloned the project, and lined it properly with the extra-components configuration.
Everything compiles fine, and uploads to the microcontroller.
However, there seems to have an issue, with I2C not working.

The output I get in the serial port is as follows:
`

    I (30) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
    I (30) boot: compile time 03:18:06
    I (31) boot: chip revision: 1
    I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
    I (41) boot.esp32: SPI Speed : 40MHz
    I (46) boot.esp32: SPI Mode : DIO
    I (50) boot.esp32: SPI Flash Size : 4MB
    I (55) boot: Enabling RNG early entropy source...
    I (60) boot: Partition Table:
    I (64) boot: ## Label Usage Type ST Offset Length
    I (71) boot: 0 nvs WiFi data 01 02 00009000 00006000
    I (78) boot: 1 phy_init RF data 01 01 0000f000 00001000
    I (86) boot: 2 factory factory app 00 00 00010000 00100000
    I (93) boot: End of partition table
    I (98) boot_comm: chip revision: 1, min. application chip revision: 0
    I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0713c ( 28988) map
    I (125) esp_image: segment 1: paddr=0x00017164 vaddr=0x3ffb0000 size=0x021f0 ( 8688) load
    I (129) esp_image: segment 2: paddr=0x0001935c vaddr=0x40080000 size=0x00404 ( 1028) load
    I (134) esp_image: segment 3: paddr=0x00019768 vaddr=0x40080404 size=0x068b0 ( 26800) load
    I (154) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x15c3c ( 89148) map
    I (189) esp_image: segment 5: paddr=0x00035c64 vaddr=0x40086cb4 size=0x042fc ( 17148) load
    I (202) boot: Loaded app from partition at offset 0x10000
    I (202) boot: Disabling RNG early entropy source...
    I (203) cpu_start: Pro cpu up.
    I (206) cpu_start: Application information:
    I (211) cpu_start: Project name: Comm-ESP-IDF
    I (217) cpu_start: App version: 1
    I (221) cpu_start: Compile time: Dec 11 2020 03:17:20
    I (227) cpu_start: ELF file SHA256: 98c72da0a3535c2a...
    I (233) cpu_start: ESP-IDF: 3.40100.200827
    I (239) cpu_start: Starting app cpu, entry point is 0x400815dc
    I (0) cpu_start: App cpu up.
    I (249) heap_init: Initializing. RAM available for dynamic allocation:
    I (256) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
    I (262) heap_init: At 3FFB2A58 len 0002D5A8 (181 KiB): DRAM
    I (268) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
    I (275) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
    I (281) heap_init: At 4008AFB0 len 00015050 (84 KiB): IRAM
    I (287) cpu_start: Pro cpu start user code
    I (306) spi_flash: detected chip: generic
    I (306) spi_flash: flash io: dio
    I (306) cpu_start: Starting scheduler on PRO CPU.
    I (0) cpu_start: Starting scheduler on APP CPU.
    E (0) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (1329) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (1829) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (2329) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (2829) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (3329) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error
    E (3829) i2c: C:\Users\ron.platformio\packages\framework-espidf\components\driver\i2c.c:726 (i2c_set_timeout):i2c timing value error

`
..etc etc

Just to clarify, a simple code on the arduino IDE compiled and uploaded for the ESP32 dev board, works fine, which means the problem is not the I2C module or the address (0x20 = 32)

`

    
    #define LCM2004A_I2C_ADR1	((uint8_t) 0x20)		// default PCF8574A I2C address
    #define LCM2004A_I2C_BL_PIN	((uint8_t) 3)			// P3: LCD backlight pin number
    #include <Wire.h>
    #include <jm_PCF8574.h>
    
    jm_PCF8574 pcf8574; // I2C address fixed later by begin(...)
    
    void setup()
    {
        Serial.begin(115200);
        while (!Serial && millis()<3000); // wait for USB Serial ready
        Serial.println();
        Serial.flush();
        Wire.begin();
    
        pcf8574.begin(LCM2004A_I2C_ADR1);
    
        pcf8574.pinMode( LCM2004A_I2C_BL_PIN, OUTPUT );
        pcf8574.digitalWrite( LCM2004A_I2C_BL_PIN, LOW );
    }
    
    void loop()
    {
        pcf8574.digitalWrite( LCM2004A_I2C_BL_PIN, HIGH );
        delay(1);
    
        pcf8574.digitalWrite( LCM2004A_I2C_BL_PIN, LOW );
        delay(1);
    }
`

RonKass, 2020-12-11:
Interesting. 
The issue comes from the i2c_set_timeout function in i2c.c (line 726), where the passed timeout (0xFFFFFF) is higher than the maximum allowed, I2C_TIME_OUT_REG_V, which is 0xFFFFF.
I'm not sure if this is an error (typo) in I2C_TIME_OUT_REG_V, where they have missed one F in the hex value, or if instead, someone added one extra F by mistake to the set timeout.

RonKass, 2020-12-11:
So.. upon further investigation..
In i2cdev.c, you are setting the timeout like this..
`

    // Timeout cannot be 0
    uint32_t ticks = dev->timeout_ticks ? dev->timeout_ticks : I2CDEV_MAX_STRETCH_TIME;
    if ((ticks != t) && (res = i2c_set_timeout(dev->port, ticks)) != ESP_OK)
        return res;`
where `I2CDEV_MAX_STRETCH_TIME` is actually set to 0xFFFFFF (six Fs)
This value set in ticks, which is passed to the i2c_set_timeout, violates the maximum allowed timeout check in i2c.c

`I2C_CHECK((timeout <= I2C_TIME_OUT_REG_V) && (timeout > 0), I2C_TIMEING_VAL_ERR_STR, ESP_ERR_INVALID_ARG);`

So the conflict is simple. I am now sure who is "wrong" here. i2c.c or i2cdev.c.
Can you have a look and tell me what you think?

UncleRus, 2020-12-13:
Thank you for reporting this! Feel free to reopen if error still occurs

commit 31620162 (2020-12-13): i2cdev: Fix #139 (clock stertching on esp32s)
closed 2020-12-13 by UncleRus