UncleRus/esp-idf-lib #140: Clock stretch timeout setting on ESP32 failing

desvdp, 2020-12-14:
**Device type**
- [x] ESP32
- [] ESP8226

**Framework version**
- [x] ESP-IDF latest master
- [] ESP-IDF 3.2.x
- [] ESP-IDF 3.3.x
- [] ESP8266 RTOS SDK latest master
- [] ESP8266 RTOS SDK 3.2

**Describe the bug**
Hi, I believe there's 2 oddities in the clock stretching for the ESP32, both in the file i2cdev.c, function i2c_setup_port 
(1) Unlike for the ESP8266 the clock stretching for the ESP32 is always performed, even if the config is equal (cond: line 150), is this intentional? 
(2) When running this code, the timeout for an ESP32 is set to ~16M (0xFFFFFF) by default. This value fails on my device; providing a lower setting fixes the init code (the read back t value -line 183- is 2000, that works).  

Some extra info about the ESP32 device/IDF I'm running:
I (30) boot: ESP-IDF v4.2-rc-5-g511965b26-dirty 2nd stage bootloader
I (30) boot: compile time 08:18:03
I (30) boot: chip revision: 1

Thanks for this great lib !

desvdp, 2020-12-14:
UPDATE: it sees the i2c driver checks against I2C_TIME_OUT_REG_V which is 0xFFFFF for ESP32 and  0xFFFFFF for ESP32S
look like #define I2CDEV_MAX_STRETCH_TIME I2C_TIME_OUT_REG_V  would be the better solution ... 

ref: https://github.com/espressif/esp-idf/search?q=%22%23define+I2C_TIME_OUT_REG_V%22

UncleRus, 2020-12-14:
See 3162016

closed 2020-12-14 by UncleRus

desvdp, 2020-12-14:
Oops, missed that commit; sorry for that.  (Good we came to the same conclusion ;-) )