UncleRus/esp-idf-lib #145: TSL2591 and IR (Rx) 38KHz, library request for both

Bytesforus, 2020-12-23:
Great work on the libraries and examples.

I am hereby request a library for the AMS TSL2591.

General link: https://ams.com/TSL25911
Datasheet: https://ams.com/documents/20143/36005/TSL2591_DS000338_6-00.pdf

And a library to read incoming IR signals (at 38KHz).

Arduino libraries exist for both of the above but I did not find an ESP-IDF one.

Kind Regards,

trombik, 2020-12-24:
https://github.com/adafruit/Adafruit_TSL2591_Library

Bytesforus, 2020-12-24:
Thanks but this an arduino library?

trombik, 2020-12-24:
sorry for the confusing post. I just wanted to leave a reference for the arduino library just for the record. I _think_ I bought a breakout board, but cannot find it. I'm not sure I can help here, but when time permits, I will.

juliandoerner, 2020-12-25:
I will port my driver for i2cdev these days.

Bytesforus, 2020-12-27:
Ok great. The IR library might be done by Espressif as they refer to its functionality in their datasheet (4.1.13).

juliandoerner, 2020-12-27:
Espressif already provides an IR library. You will find more information about the RMT (Remote controll) under peripherals/rmt.

Bytesforus, 2020-12-27:
Ok great, I searched for IR only... Thanks.

closed 2020-12-29 by UncleRus