UncleRus/esp-idf-lib #146: Abstraction layer for SPI

UncleRus, 2020-12-25:
We need a library that implements a hardware abstraction layer for SPI devices, something like i2cdev.
There are a lot of good devices that we can support.

Bytesforus, 2021-01-09:
https://github.com/loboris/ESP32_TFT_library

UncleRus, 2021-03-26:
It looks like, unlike I2C, different devices use the SPI protocol in different ways, and it will be extremely difficult to develop and maintain a shared SPI transport library for ESP8266 and ESP32. I think it's best to leave the current approach for now: implement SPI separately for ESP8266 components only if necessary.

UncleRus, 2021-03-26:
I'm closing this issue for now, but feel free to reopen it in case of good ideas.

closed 2021-03-26 by UncleRus