UncleRus/esp-idf-lib #150: i2cdev.h in relation to the Espressif included i2c.h driver?

Bytesforus, 2020-12-27:
Several questions.

Can i2cdev.h cause conflicts with the main Espressif i2c.h driver if used simultaneously?

Is it helpfull and or necessary to double the i2c command structure like this?

And what is the motivation behind the code layering employed by a separate .c file, like in most examples:

  main.c -> The example code.
  ds3231.h -> The header library as referred to in main.c declares functions found in ds3231.c but it does not include ds3231.c?
  ds3231.c -> However ds3231.c itself includes ds3231.h, which is essentially a back-referral?

Why not command the IC directly in main.c using the i2c.h driver? Are there (timing, sequence or other) issues to be expected on the ESP32 when doing so?

  main.c including i2c.h

  instead of:

  main.c -> some.h <-- another.c --> i2cdev.c --> i2cdev.h --> i2c.h <-- i2c.c

The library style used is, in part, employed due to (other ESP-IDF libraries employing) freeRTOS?

I need to consider the additional use of memory by unused functionality as most of these ESP32 units are only 4MB.

Stripping code to meet the minimum required functionality and checking the relevant logic becomes a bigger challenge this way.

I am new to the IDF and freeRTOS so I appreciate your comments. Thanks again for the work done.

juliandoerner, 2020-12-27:
If used wisely, i2cdev.h and i2c.h driver wont cause conflicts if used simultaneously. The whole point about writing an abstraction layer like i2cdev.h is thread safety. The provided drives in i2c.h are not thread safe, as mentioned in the esp-idf documentation. Calling the standard i2c driver from different tasks will cause problems and one has to come up with something to synchronize them. The pattern found in i2cdev.h is a common pattern to do this. 

You will found several information about the code layering in c on the web. It is not specific to this library.

Why is not everything in the main.c? Because this library is about reuse. One can easily put necessary component to its own project and use it.

The library style is indeed employed because of freeRTOS. Like mentioned, the power to do stuff in parallel needs special care, hence this library.

The additional use of memory shouldn't be a problem. 4MB is quite a lot and the compiler will of cause only use the necessary functionality for your specific use case. The abstraction of certain things is important in programming. Writing the same i2c driver for every new project will cause the same problems again and again.

Bytesforus, 2020-12-27:
Thanks for your patience. I am constantly reading up on all the material, Espressif and otherwise.

The i2c.h is not thread safe by itself but it is utilized by the i2cdev.h in a thread safe manner for use in freeRTOS, ok got it. But thread safety or priority can also be obtained through process markers and timers, when not using freeRTOS? I currently doubt freeRTOS use for some applications, although I likely can not escape it. At times it seems complex for its intended purpose?

Like how does ds3231.h know about the existence of ds3231.c? Let us say I want to edit the content of the "ds3231_get_time" function, how do I learn that the ds3231.c file even exists and contains this function? Also in relation to changing i2c frequencies.

A problem might be that functionality is somewhat hiden and/or is coded more than once. You can use the necessary component in your project with ease but you still need to check all of the remaining code (files) to see if you meet all the requirements.

I do not yet fully grasp all the functionality in freeRTOS or the use of "C" code layering in general so bear with me here.

It is indeed helpful to learn that the compiler does not load the unused parts of the code, then 4MB should be fine.

Bytesforus, 2021-01-02:
I now understand that the compiler automatically finds the 3231.c file when it is located in the same directory.

Still, freeRTOS has its challenges but I need to learn how to deal with those. Thanks again, question closed.

closed 2021-01-02 by Bytesforus