UncleRus/esp-idf-lib #162: Driver for SIMCOM SIM7600 series

DhruvaG2000, 2021-03-09:
I am requesting a driver for `SIM7600X` series.

### Device description

* The `SIM7600` is Multi-Band LTE-TDD/LTE-FDD/HSPA+ and GSM/GPRS/EDGE module solution in a SMT type which supports LTE CAT1 up to 10Mbps for downlink data transfer.
* It has strong extension capability with rich interfaces including UART, USB2.0, I2C, GPIO etc. With abundant application capability like TCP/UDP/FTP/FTPS/HTTP/HTTPS/DNS, the module provides much flexibility and ease of integration for customer's application.

### Device information

The SIM7600 series integrates multiple satellite high accuracy positioning GNSS systems, with multiple built-in network protocols, supports drivers for main operation systems (USB driver for Windows, Linux and Android etc.) and software function, AT commands are compatible with SIM7500/SIM7600-H series modules.  Meanwhile the SIM7600 series integrates main industrial standard interfaces, with powerful expansibility, including abundant interfaces such as UART, USB, I2C, GPIO, which is suitable for main IoT applications such as telematics, surveillance devices, CPE, industrial routers, and remote diagnostics etc. 

The product information and the data sheet can be found at:
https://www.simcom.com/product/SIM7600X.html 

### Purchasing the product

The device can be purchased at:

- [ ] [Rhydolabz](https://www.googleadservices.com/pagead/aclk?sa=L&ai=DChcSEwj6uvzs0KLvAhXpnUsFHa3UDPIYABAGGgJzZg&ohost=www.google.com&cid=CAESQOD28hpyymucKrrBG18oe-Y6e3oL_UkqNw1D4SeSA4gmwJKNoo063FGz8h44syN7zGgX9v5aOr5b05ny9fwSC7Q&sig=AOD64_0rmyk7Awlx5aLUOKi-9bIHFApNdQ&ctype=5&q=&ved=2ahUKEwi53vLs0KLvAhXWZCsKHVK8DFUQ9aACegQIBBBi&adurl=)

### Other implementation

There is a Arduino GSM library  at https://github.com/vshymanskyy/TinyGSM

DarkExodus04, 2021-03-09:
Yes,this would be useful.

UncleRus, 2021-03-10:
This will take some time. Firstly, I don't have this module and secondly, I can't just port TinyGSM because of the license: only licenses allowed in this project are MIT, BSD, EPL, ISC, CC and so on, no GPL/LGPL.

UncleRus, 2021-03-10:
But, of course, you can develop the driver yourself or at least help me (or other developers) in debugging.

DhruvaG2000, 2021-03-10:
Yes, let us know how we can stay in touch, We would be happy to test, debug and develop.

JaimeAlbq, 2022-10-11:
Hello @olliiiver,

I saw that you have created a driver for [SIM7600](https://github.com/olliiiver/esp_lte_modem).

Would it be possible to help us porting it, since it is MIT licensed, to this library?

txf-, 2024-11-15:
> I am requesting a driver for `SIM7600X` series.

Have you considered using esp-modem: https://github.com/espressif/esp-protocols/tree/32ac21b03ce831178a8142c41801640de26f045c/components/esp_modem

trombik, 2025-08-01:
moved to: https://github.com/esp-idf-lib/core/issues/16

closed 2025-08-01 by trombik