UncleRus/esp-idf-lib #164: Support for tgsong / amazon-freertos-examples  FreeRTOS AWS ESP32

Yoimer, 2021-03-09:
@UncleRus Hello. Amazing piece of work you've made. Congratulations. Is there any way your set of libraries could be implemented on this project [tgsong](https://github.com/tgsong/amazon-freertos-examples). Looking forward to hearing your comments. Thanks in advance.

UncleRus, 2021-03-10:
Sorry, I don't understand the question. Do you want to use esp-idf-lib in your examples? Of course, you can do this, just add it as a submodule to your repository and use it as written in README.

mbuckaway, 2021-03-11:
I'll add my two cents. I use the ESP AWS MQTT lib from Espressif to connect to my ESP32 devices to the AWS IoT Core. Espressif has done all the hard work of porting the AWS lib for you. No need to fight with Amazon's version of FreeTOS. THis allows you to make easy use of the ESP-IDF-LIB or any other ESP-IDF component. See the Espressif Github for details.

Yoimer, 2021-03-11:
@UncleRus @mbuckaway Thanks for replying.

So @UncleRus. I was able to use your DHT library in my current AWS FreeRTOS ESP32 project (not an easy task due to AWS way to configure FreeRTOS)

@mbuckaway It's imperative to me using AWS FreeRTOS ESP32

UncleRus, 2021-03-11:
I’m very sorry, but unfortunately I don’t plan to support AWS FreeRTOS. However, I will be happy if you implement this support.

Yoimer, 2021-03-15:
@UncleRus Hello there. I was able to implement only the `DHT11` library. Even in the very native .zip file for `FreeRTOS V202012.00`. The rest of the libraries, I could not.

closed 2021-04-12 by UncleRus