UncleRus/esp-idf-lib #187: Requesting SSD1306 Oled Driver with i2c using ESP-idf

chandanchan0086, 2021-04-28:
I am requesting a driver for `SSD1306`.

### Device description

The SSD1306 is a 128×64 dot single chip driver with a controller that's used for graphic display systems. It's commonly integrated into OLED display modules.

We can control this by using I2c communication and this is used for display purposes.

`SSD1306` is a chip to do a display. It supports I2C. It
supports VDD from 3.3V to 5V.

### Device information

The product information and the data-sheet can be found at : https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf

### Purchasing the product

This device is purchased at Amazon.

The URLs where you can buy the device are:

https://www.amazon.in/Generic-0-96-Display-Module-Arduino/dp/B01GNZEM22/ref=sr_1_2?adgrpid=59355583142&dchild=1&ext_vrnc=hi&gclid=CjwKCAjwj6SEBhAOEiwAvFRuKFY1c1huvszWyY-__zIDV7D79wkzN8Lb3oENsAPsoMssbyD9ILtlhxoCClEQAvD_BwE&hvadid=398044508860&hvdev=c&hvlocphy=20460&hvnetw=g&hvqmt=b&hvrand=2518053099860831822&hvtargid=kwd-296468185891&hydadcr=7760_1981943&keywords=esp8266+oled&qid=1619612738&sr=8-2

UncleRus, 2021-04-28:
Sorry, but I have to say no: there will be no display drivers in this repository. 

The reasons have been mentioned many times, here is a recent example: #177

closed 2021-04-28 by UncleRus

chandanchan0086, 2021-04-28:
Okay and thank you for your quick response.  Actually im facing same problem like #74 so i have requested you for ssd1306 library. 

Can you please help me out from this? How to resolve this? Like i have to read values from qmc5883l magneto meter (i have used your library for this) and i need to display those values on ssd1306 oled display ( for this I have used library from this https://github.com/TaraHoleInIt/tarablessd1306). 

But my code is crashing ...!!!.

Can you please help me out with this?

UncleRus, 2021-04-28:
> my code is crashing

Please create a separate issue and post `idf.py monitor` output so we can see where it fails.

UncleRus, 2021-04-28:
We also need to see your code because all of our mind readers are on vacation.