UncleRus/esp-idf-lib #188: ESP crashes while using qmc5883l with ssd1306 oled

chandanchan0086, 2021-04-28:
**Device type**
    ESP32

**Framework version**
- [x] ESP-IDF latest master

**Describe the bug**
I have to read values from the qmc5883l magnetometer (i have used https://github.com/UncleRus/esp-idf-lib/tree/master/components/qmc5883l library for this) and I need to display those values on ssd1306 OLED display ( for this I have used library from this https://github.com/TaraHoleInIt/tarablessd1306).

Both are different libraries and When I try to use both in the same code, ESP crashes.

idf.py monitor output:- 
[image]

program:-

#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include <stdbool.h>
#include "ssd1306.h"
#include "ssd1306_draw.h"
#include "ssd1306_font.h"
#include "ssd1306_default_if.h"

#include "qmc5883l.h"
#include <math.h>
#define PI 3.142857
#define PIN_SDA 5
#define PIN_CLK 4
qmc5883l_t dev;
#define Declination       -1.18333 //http://www.magnetic-declination.com 

static const int I2CDisplayAddress = 0x3C;
static const int I2CDisplayWidth = 128;
static const int I2CDisplayHeight = 64;
static const int I2CResetPin = -1;

struct SSD1306_Device I2CDisplay;

double Heading;

bool DefaultBusInit( void ) {
    assert( SSD1306_I2CMasterInitDefault( ) == true );
    assert( SSD1306_I2CMasterAttachDisplayDefault( &I2CDisplay, I2CDisplayWidth, I2CDisplayHeight, I2CDisplayAddress, I2CResetPin ) == true );
    return true;
}

static void wait_for_data()
{
    while (true)
    {
        bool ready;
        ESP_ERROR_CHECK(qmc5883l_data_ready(&dev, &ready));
        if (ready)
            return;
        vTaskDelay(1);
    }
}

//compass initialization
void compass_init(){
	memset(&dev, 0, sizeof(qmc5883l_t));
    	ESP_ERROR_CHECK(i2cdev_init());
	ESP_ERROR_CHECK(qmc5883l_init_desc(&dev, 0, QMC5883L_I2C_ADDR_DEF, (gpio_num_t)PIN_SDA, (gpio_num_t)PIN_CLK));
	ESP_ERROR_CHECK(qmc5883l_set_mode(&dev, QMC5883L_MODE_CONTINUOUS));
	//ESP_ERROR_CHECK(qmc5883l_get_chip_id(&dev, &id));
	//printf("\nID - %d", id);
   	 // 50Hz data rate, 128 samples, -2G..+2G range
    	ESP_ERROR_CHECK(qmc5883l_set_config(&dev, QMC5883L_DR_50, QMC5883L_OSR_128, QMC5883L_RNG_2));
	//ESP_ERROR_CHECK(qmc5883l_set_int(&dev, false));
}

//compass function to read values
void compass(){
	wait_for_data();
	qmc5883l_data_t data;
        if (qmc5883l_get_data(&dev, &data) == ESP_OK){
            printf("Magnetic data: X:%.2f mG, Y:%.2f mG, Z:%.2f mG\n", data.x, data.y, data.z);
            Heading = atan2((double)data.y, (double)data.x) + Declination;
  	if (Heading>2*PI) /* Due to declination check for >360 degree */
   		Heading = Heading - 2*PI;
  	if (Heading<0)    /* Check for sign */
   		Heading = Heading + 2*PI;
  	printf("Heading = %f\n",Heading* 180 / PI);  // Convert into angle and return
            }
        else
            printf("Could not read QMC5883L data\n");
	vTaskDelay(1/portTICK_PERIOD_MS);
}

void app_main(void)
{
	char stt[100];
    	if ( DefaultBusInit( ) == true ) {
        	printf( "BUS Init lookin good...\n" );
   	 }
   	 compass_init();
        while(1)
        {
		SSD1306_Clear( &I2CDisplay, SSD_COLOR_BLACK );
		SSD1306_SetFont( &I2CDisplay, &Font_droid_sans_fallback_11x13 );
		compass();
		sprintf(stt, "%.2lf", Heading);
		SSD1306_FontDrawString( &I2CDisplay, 0,0, stt, SSD_COLOR_WHITE );
		SSD1306_Update( &I2CDisplay );
		vTaskDelay(500 / portTICK_RATE_MS);
       }
}

UncleRus, 2021-04-28:
1. SSD1306 and qmc5883l must use _different_ I2C ports. For example, SSD1306 by default on port 0 and qmc5883l on port 1:
```C
qmc5883l_init_desc(&dev, 1, QMC5883L_I2C_ADDR_DEF, (gpio_num_t)PIN_SDA, (gpio_num_t)PIN_CLK));
```
2. Init i2cdev and qmc5883l first and then SSD1306.

UncleRus, 2021-04-29:
So common template for using i2cdev with non-i2cdev components is:

1. Initialize i2cdev with `i2cdev_init()`
2. Initialize device descriptors which drivers based in i2cdev on I2C port 1
3. Initialize any other library on I2C port 0.

I'm closing this issue, feel free to reopen if you have any questions again.

closed 2021-04-29 by UncleRus

chandanchan0086, 2021-04-30:
@UncleRus Thank you so much for your time and help and this issue is cleared.