UncleRus/esp-idf-lib #202: How to use TCA9548 I2C Multiplexer library

ameerhazo, 2021-05-07:
Hello everyone, I'm trying to use the TCA9548 I2C Multiplexer library to use 8 different sensors on its 8 different buses. To test it I've managed to initiate its mutex and set its channel to one of its bus which has a BME680 sensor attached to it. Unfortunately, I cant seem to talk to the BME680 on the channel because the sensor has a different address than the I2C Mux and obviously, the ESP32 cant find the sensor. Ive used A I2C Mux before in the Arduino environment and after setting the channel, I'm able to do a sensor.begin() for the sensor. Should I then change the address of the sensor in the sensor_init_desc() to the address of the I2C Mux? Would it not return an error for the init because the address does not correspond to the addresses of the sensors in the header file?  

Thanks for the help guys.

Ameer

ameerhazo, 2021-05-07:
So the solution is to initiate the sensors with the address of the I2C Mux. I had to change the checks in the init to allow the mux address. Would be nice if the 7 mux address can be added in the init aswell in case someone uses the sensors with a mux.

closed 2021-05-07 by ameerhazo

UncleRus, 2021-05-07:
> 
    AMG8833 Grid-Eye Thermal Cam
    TCS34725 Color Sensor
    LIS3DH 3-Axis Accelerometer
    AS3935 Lightning Detector
    VL53L1x Laser Distance Sensor (not finished testing yet).

Wow! That would  be great! Please see [CONTRIBUTING.md](https://github.com/UncleRus/esp-idf-lib/blob/master/CONTRIBUTING.md) in this repository.