UncleRus/esp-idf-lib #203: request a driver for `MAX31865`.

zgjsntsm, 2021-05-07:
I am requesting a driver for `MAX31865`.

### Device description

The MAX31865 is an easy-to-use resistance-to-digital
converter optimized for platinum resistance temperature
detectors (RTDs). . It support SPI. It
supports VDD from 3.0V to 3.6V.

### Device information

The product information and the data sheet can be found at:

https://datasheets.maximintegrated.com/en/ds/MAX31865.pdf

### Other implementation

There is a driver for Amiga 1000 at https://github.com/adafruit/Adafruit_MAX31865

UncleRus, 2021-06-20:
@zgjsntsm Help me with testing #221

limpens, 2021-07-21:
@UncleRus 

I tested the max31865 example code, it continuously throws an error:

```
I (291) spi_flash: detected chip: generic
I (295) spi_flash: flash io: dio
I (300) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (70) max31865: [CS 5] Fault detected
E (70) max31865-example: Failed to measure: -1 (ESP_FAIL)
E (340) max31865: [CS 5] Fault detected
E (340) max31865-example: Failed to measure: -1 (ESP_FAIL)
E (610) max31865: [CS 5] Fault detected

```

For educational purposes I set CS to -1, no errors, but no valid temperature readings.
Same hardware is working properly with the Adafruit MAX31865 library when using the Arduino framework

limpens, 2021-07-22:
@UncleRus 

When the [spi mode](https://github.com/UncleRus/esp-idf-lib/blob/max31865/components/max31865/max31865.c#L151) is changed to 1 **and** a cs_ena_pretrans=1 is introduced, the readings look fine to me.

```
     dev->spi_cfg.spics_io_num = cs_pin;
     dev->spi_cfg.clock_speed_hz = CLOCK_SPEED_HZ;
+    dev->spi_cfg.mode = 1;
-    dev->spi_cfg.mode = 0;
     dev->spi_cfg.queue_size = 1;
+    dev->spi_cfg.cs_ena_pretrans = 1;
     //dev->spi_cfg.flags = SPI_DEVICE_NO_DUMMY;
```

I don't know if this has any negative side effects. The [Espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#_CPPv4N29spi_device_interface_config_t15cs_ena_pretransE) note the pretrans is for half-duplex transactions

Regards, Eric.

UncleRus, 2021-07-24:
@limpens Thank you!

UncleRus, 2021-07-26:
#221 merged

closed 2021-07-26 by UncleRus