UncleRus/esp-idf-lib #206: CI: clang_format is waaaaay too strict

UncleRus, 2021-05-10:
See #205 

First of all, 80 columns? Why? Modern monitors usually shows more than 80 columns of text without problems.

Why am I not allowed to align values vertically? It's good for reading code:
```C
static int comething[] = {
    [SHORT_IDX]       = 10,
    [VERY_LONG_INDEX] = 20
};
```

I prefer to use open brace in enum typedefs at the same line:
```C
typedef enum {
    BUTTON_PRESSED = 0,
    BUTTON_RELEASED,
} something_t;
```

```
--- a/examples/button/main/main.c
+++ b/examples/button/main/main.c
@@ -1,15 +1,15 @@
-#include <stdio.h>
-#include <esp_log.h>
-#include <esp_idf_lib_helpers.h>
 #include <button.h>
+#include <esp_idf_lib_helpers.h>
+#include <esp_log.h>
+#include <stdio.h>
```
What? Why? There is nothing wrong in placing headers this way.

I want to place short `if` in one line: 
```C
if (res != ESP_OK) break;
```

And so on.

To summarize this:
As it stands, checking code formatting hinders rather than helps. I believe that rules are for convenience, not for suffering.
I've disabled clang_format in CI.

trombik, 2021-05-10:
if you don't like the style, please change it. your repository, your rules.

I cannot answer most of the questions here because I do not know how these style rules were justified. I actually do not want to know various justifications because it is not productive. some are based on valid technical reasons, others are on preferences, or very subjective. LLVM style was chosen because it looks similar to the existing one, and some rules, not enforced by `clang-format`, were added from `KNF` because rules should be explicitly stated even if it is obvious to many. none of them are my rule.

to me, the whole point of code style, or rules in a project, is to discipline the project, and the project members.  i was in a project where most of commit messages were "fixed a bug", "oops", or something similar. the code style was "no style". the code has no patterns. as members join and leave, the code got very inconsistent. as it was very hard to review the code, nobody reviewed other's code. the practice continued until someone eventually say "we cannot maintain the code. let's rewrite from scratch", and it repeated. my opinion is, a project should choose a style, or rules, and stick with it. however, the rule should fit to the project, and it is very hard to write good rules from the beginning. there would be corner-cases, and limitations. the project and the rules should evolve together.

the CI was written because I would like you to focus on writing code. I'm sorry if it gets in your way, and it is not my intention. every project has its own rule, and I always follow whatever it is. as you write most of the code, the style should be what you are comfortable with.

UncleRus, 2021-05-11:
I perfectly understand your reasons regarding the code style. But for twenty-five years of programming in C, I have developed my own and I absolutely do not want to spend hours configuring the clang_format settings, adjusting it to my style. In this repository, the easiest way to control the style of the sources is by me personally, without using automation. The same goes for checking markdown files.
I remove C and markdown style checks from the CI.

commit 999be489 (2021-05-11): CI: remove linting, fixes #206