UncleRus/esp-idf-lib #210: SHT4X Invalid CRC

SERIDJ, 2021-05-19:
**Device type**:

- [x] ESP32

**Framework version**:

- [x]  ESP-IDF v4.1-332-g7b7c64107-dirty 2nd stage bootloader

**Describe the bug**:

hi, i try to test SHT4X and it not work with this library ,bcs i tested it with arduino c++ code it work well.

TERMINAL output : 

```
I (269) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (275) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (281) heap_init: At 4008AF8C len 00015074 (84 KiB): IRAM
I (288) cpu_start: Pro cpu start user code
I (306) spi_flash: detected chip: gd
I (306) spi_flash: flash io: dio
W (306) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
master.clk_speed = 100000
E (2322) i2cdev: Could not read from device [0x44 at 0]: 263
E (2322) sht4x: Invalid CRC
ESP_ERROR_CHECK failed: esp_err_t 0x109 (ESP_ERR_INVALID_CRC) at 0x400848dc
file: "../main/main.c" line 109
func: app_main
expression: sht4x_init(&dev)

ELF file SHA256: 52bda6be7360e7c8

Backtrace: 0x40084321:0x3ffb4710 0x400848df:0x3ffb4730 0x400d3d64:0x3ffb4750 0x400d0cfa:0x3ffb4780 0x4008591d:0x3ffb47a0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7040
ho 0 tail 12 room 4
load:0x40078000,len:13100
load:0x40080400,len:3884
entry 0x40080688

```

and i cant run SGP40 VOC module too ...

UncleRus, 2021-05-19:
Thank you for testing, it's extremely valuable to us.
However, I kindly ask you to create one issue per bug.

sht4x error is not CRC-based. This is
```
i2cdev: Could not read from device [0x44 at 0]: 263
```
263 = ESP_ERR_TIMEOUT
In other words, we just can't get correct response from device to the serial number request.
It will be extremely helpful if you have a logic analyzer and use it to capture transactions between ESP and SHT.

UncleRus, 2021-05-19:
SGP40 continuation: #211

SERIDJ, 2021-05-19:
@UncleRus  oky , i dont have hardware analyzer , and i dont know if i can use software to capture  transactions between ESP and SHT. i will search ...

UncleRus, 2021-05-20:
Try now, I added 10ms timeout to read serial command

SERIDJ, 2021-05-20:
@UncleRus  ah good it work well now ,

```
 (0) cpu_start: App cpu up.
I (249) heap_init: Initializing. RAM available for dynamic allocation:
I (256) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (262) heap_init: At 3FFB28E8 len 0002D718 (181 KiB): DRAM
I (269) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (275) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (281) heap_init: At 4008AF8C len 00015074 (84 KiB): IRAM
I (288) cpu_start: Pro cpu start user code
I (306) spi_flash: detected chip: gd
I (306) spi_flash: flash io: dio
W (306) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
sht4x Sensor: 23.47 Â°C, 36.61 %
sht4x Sensor: 23.45 Â°C, 36.62 %
sht4x Sensor: 23.47 Â°C, 36.57 %
sht4x Sensor: 23.46 Â°C, 36.59 %
sht4x Sensor: 23.44 Â°C, 36.61 %
sht4x Sensor: 23.46 Â°C, 36.62 %
sht4x Sensor: 23.45 Â°C, 36.61 %
sht4x Sensor: 23.45 Â°C, 36.63 %
sht4x Sensor: 23.44 Â°C, 36.63 %
sht4x Sensor: 23.44 Â°C, 36.63 %

```

closed 2021-05-20 by UncleRus

SERIDJ, 2021-05-20:
@UncleRus  i am not sure but i test sht3x and i think it have same bug....

UncleRus, 2021-05-20:
No, I don't think so: this is writing to device, not reading from it.
An i2c write timeout means the slave device doesn't respond with an ACK.

Please create a new issue for this.

SERIDJ, 2021-05-20:
its work , not need to add an issue,

thanks ,