UncleRus/esp-idf-lib #211: sgp40:  ESP_ERR_INVALID_ARG (0x102)

UncleRus, 2021-05-19:
**Device type**:

- [x] ESP32
- [x] ESP8226

**Framework version**:

- [x] ESP-IDF latest master
- [x] ESP-IDF 3.2.x
- [x] ESP-IDF 3.3.x
- [x] ESP8266 RTOS SDK latest master
- [x] ESP8266 RTOS SDK 3.2

Follow up of #210 

```
I (263) heap_init: At 3FFB28A8 len 0002D758 (181 KiB): DRAM
I (269) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (276) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (282) heap_init: At 4008AFE8 len 00015018 (84 KiB): IRAM
I (288) cpu_start: Pro cpu start user code
I (306) spi_flash: detected chip: gd
I (307) spi_flash: flash io: dio
W (307) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (317) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400848dc
file: "../main/main.c" line 65
func: task
expression: sgp40_init(&sgp)

ELF file SHA256: 5edb899531acb661

Backtrace: 0x40084321:0x3ffb7d60 0x400848df:0x3ffb7d80 0x400d3e65:0x3ffb7da0 0x4008591d:0x3ffb7e90

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7040
ho 0 tail 12 room 4
load:0x40078000,len:13100
load:0x40080400,len:3884
entry 0x40080688
```

SERIDJ, 2021-05-19:
```
/**
 * Simple example with humidity sensor and SGP40
 */
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_system.h>
#include <sht3x.h>
#include <sgp40.h>
#include <string.h>
#include <esp_err.h>

/* float is used in printf(). you need non-default configuration in
 * sdkconfig for ESP8266, which is enabled by default for this
 * example. see sdkconfig.defaults.esp8266
 */

#if defined(CONFIG_IDF_TARGET_ESP8266)
#define SDA_GPIO 4
#define SCL_GPIO 5
#else
#define SDA_GPIO 21
#define SCL_GPIO 22
#endif

#if defined(CONFIG_IDF_TARGET_ESP32S2)
#define APP_CPU_NUM PRO_CPU_NUM
#endif

#define SHT_ADDR SHT3X_I2C_ADDR_GND

static const char *voc_index_name(int32_t voc_index)
{
    if (voc_index <= 0) return "INVALID VOC INDEX";
    else if (voc_index <= 10) return "unbelievable clean";
    else if (voc_index <= 30) return "extremely clean";
    else if (voc_index <= 50) return "higly clean";
    else if (voc_index <= 70) return "very clean";
    else if (voc_index <= 90) return "clean";
    else if (voc_index <= 120) return "normal";
    else if (voc_index <= 150) return "moderately polluted";
    else if (voc_index <= 200) return "higly polluted";
    else if (voc_index <= 300) return "extremely polluted";

    return "RUN!";
}

void task(void *pvParamters)
{
    //sht3x_t sht;
    sgp40_t sgp;

 /*
    // setup SHT3x
    memset(&sht, 0, sizeof(sht));
    ESP_ERROR_CHECK(sht3x_init_desc(&sht, 0, SHT_ADDR, SDA_GPIO, SCL_GPIO));
    ESP_ERROR_CHECK(sht3x_init(&sht));
    // Start periodic measurements with 2 measurements per second.
    ESP_ERROR_CHECK(sht3x_start_measurement(&sht, SHT3X_PERIODIC_2MPS, SHT3X_HIGH));
    printf("!!!! Humidity sensor initilalized\n");
*/
    // setup SGP40
    memset(&sgp, 0, sizeof(sgp));
    ESP_ERROR_CHECK(sgp40_init_desc(&sgp, 0, SDA_GPIO, SCL_GPIO));
    ESP_ERROR_CHECK(sgp40_init(&sgp));
    printf("!!!! SGP40 initilalized. Serial: %04X %04X %04X\n",
            sgp.serial[0], sgp.serial[1], sgp.serial[2]);

    // Wait until all set up
    vTaskDelay(pdMS_TO_TICKS(250));

    TickType_t last_wakeup = xTaskGetTickCount();
    while (1)
    {
        // Get the RHT values
        float temperature=25, humidity=50;
        //ESP_ERROR_CHECK(sht3x_get_results(&sht, &temperature, &humidity));

        // Feed it to SGP40
        int32_t voc_index;
        ESP_ERROR_CHECK(sgp40_measure_voc(&sgp, humidity, temperature, &voc_index));

        printf("!!!! %.2f Â°C, %.2f %%, VOC index: %d, Air is [%s]\n",
                temperature, humidity, voc_index, voc_index_name(voc_index));

        // Wait until 1 seconds (VOC cycle time) are over.
        vTaskDelayUntil(&last_wakeup, pdMS_TO_TICKS(1000));
    }
}

void app_main()
{
    ESP_ERROR_CHECK(i2cdev_init());

    xTaskCreate(task, "sgp40", configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL);
}

```

SERIDJ, 2021-05-19:
ok

SERIDJ, 2021-05-19:
@UncleRus  work well, thanks , whats was the bug ? 

```
I (21923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (22923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (23923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (24923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (25923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (26923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (27923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (28923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (29923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (30923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (31923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (32923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (33923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (34923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (35923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (36923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (37923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (38923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (39923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (40923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (41923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (42923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (43923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (44923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (45923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 0, Air is [INVALID VOC INDEX]
I (46923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 1, Air is [unbelievable clean]
I (47923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 1, Air is [unbelievable clean]
I (48923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 10, Air is [unbelievable clean]
I (49923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 30, Air is [extremely clean]
I (50923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 47, Air is [higly clean]
I (51923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 63, Air is [very clean]
I (52923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 76, Air is [clean]
I (53923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 86, Air is [clean]
I (54923) sgp40-example: 25.00 Â°C, 50.00 %, VOC index: 94, Air is [normal]

```

UncleRus, 2021-05-19:
It was lack of device for testing the driver :)

closed 2021-05-19 by UncleRus