UncleRus/esp-idf-lib #214: SCD4X CO2 :  263 (ESP_ERR_TIMEOUT)

SERIDJ, 2021-05-26:
**Device type**:

- [x] ESP32

**Framework version**:

- [x] ESP-IDF  v4.1

**Describe the bug**:

I try to test co2 sensor SCD4X with my esp32 and it work  good only with delay 5 SEC in loop, if i try with 4 SEC i got same time errors  : 
```
I (70700) scd4x-example: CO2: 1179 ppm
I (70700) scd4x-example: Temperature: 24.59 Â°C
I (70700) scd4x-example: Humidity: 37.65 %
I (74710) scd4x-example: CO2: 1178 ppm
I (74710) scd4x-example: Temperature: 24.54 Â°C
I (74710) scd4x-example: Humidity: 37.85 %
**E (79720) i2cdev: Could not read from device [0x62 at 0]: 263
E (79720) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)**
I (83720) scd4x-example: CO2: 1170 ppm
I (83720) scd4x-example: Temperature: 24.39 Â°C
I (83720) scd4x-example: Humidity: 38.07 %
**E (88730) i2cdev: Could not read from device [0x62 at 0]: 263
E (88730) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)**
I (92730) scd4x-example: CO2: 1171 ppm
I (92730) scd4x-example: Temperature: 24.32 Â°C
I (92730) scd4x-example: Humidity: 38.24 %

```

and if i try with 100ms delay i got more errors its logic...

```
I (550) scd4x-example: Sensor initialized
I (550) scd4x-example: Sensor serial number: 0x377a1f073b50
I (550) scd4x-example: Periodic measurements started
E (1660) i2cdev: Could not read from device [0x62 at 0]: 263
E (1660) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (2760) i2cdev: Could not read from device [0x62 at 0]: 263
E (2760) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
I (2860) scd4x-example: CO2: 1240 ppm
I (2860) scd4x-example: Temperature: 27.03 Â°C
I (2860) scd4x-example: Humidity: 33.80 %
E (3970) i2cdev: Could not read from device [0x62 at 0]: 263
E (3970) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (5070) i2cdev: Could not read from device [0x62 at 0]: 263
E (5070) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (6170) i2cdev: Could not read from device [0x62 at 0]: 263
E (6170) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (7270) i2cdev: Could not read from device [0x62 at 0]: 263
E (7270) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
I (7370) scd4x-example: CO2: 1365 ppm
I (7370) scd4x-example: Temperature: 26.93 Â°C
I (7370) scd4x-example: Humidity: 34.78 %
E (8480) i2cdev: Could not read from device [0x62 at 0]: 263
E (8480) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (9580) i2cdev: Could not read from device [0x62 at 0]: 263
E (9580) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (10680) i2cdev: Could not read from device [0x62 at 0]: 263
E (10680) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
E (11780) i2cdev: Could not read from device [0x62 at 0]: 263
E (11780) scd4x-example: Error reading results 263 (ESP_ERR_TIMEOUT)
I (11880) scd4x-example: CO2: 1403 ppm
I (11880) scd4x-example: Temperature: 26.73 Â°C
I (11880) scd4x-example: Humidity: 35.19 %
```

i just to know if its hardware need 5 SEC to read correctly or its soft bugs ?

THANKS.

UncleRus, 2021-05-26:
Yes, this is hardware limitation.
See datasheet.

closed 2021-05-26 by UncleRus

SERIDJ, 2021-05-26:
ok thanks