UncleRus/esp-idf-lib #218: bme measurement already running loop

ameerhazo, 2021-06-11:
**Device type**:

ESP32

**Framework version**:

- [x] ESP-IDF latest master

**Describe the bug**:

bme680 goes on a measurement already running loop. After starting the program, the task logs out "no new data" before going on an endless loop of no readings.

UncleRus, 2021-06-12:
Hmmmm...
You got "No new data" message when bit 7 ("New data flag") in register 0x1D is low, and yes, that means BME680 has no new measurement data in output registers since the last reading.
Second message means that you have bit 5 ("Measuring status flag") in the same register high, according to the datasheet the high value of this bit shows us that the data conversion is still running.

It looks like threre are no errors in driver code. But I heard of bugs like this with BME680 and one of the workaronds was to lower the I2C frequency.

And please show the actual code you using.

ameerhazo, 2021-06-12:
Hello @UncleRus, thanks for your reply to my question. I'm using the code provided in the example folder. I'll try lowering the I2C frequency and see if solves my problem. Thank you and have a nice weekend.

ameerhazo, 2021-06-16:
Hi @UncleRus, so ive lowered the i2c frequency in the bme680 source code from 400kHz to 200kHz and I'm still having the same no measurement loop. 

This is my code, the setting are as follow:

void setupBME680(void)
{
    memset(&env_sensor, 0, sizeof(bme680_t));
    bme680_init_desc(&env_sensor, BME680_I2C_ADDR_1, 0, GPIO_NUM_23, GPIO_NUM_22);
    bme680_init_sensor(&env_sensor);
    // Changes the oversampling rates to 4x oversampling for temperature
    // and 2x oversampling for humidity. Pressure measurement is skipped.
    bme680_set_oversampling_rates(&env_sensor, BME680_OSR_4X, BME680_OSR_4X, BME680_OSR_2X);
    // Change the IIR filter size for temperature and pressure to 7.
    bme680_set_filter_size(&env_sensor, BME680_IIR_SIZE_7);
    // Change the heater profile 0 to 200 degree Celsius for 100 ms.
    bme680_set_heater_profile(&env_sensor, 0, 200, 100);
    bme680_use_heater_profile(&env_sensor, 0);
    // Set ambient temperature to 10 degree Celsius
    bme680_set_ambient_temperature(&env_sensor, 18);
    bme680_free_desc(&env_sensor);
    ESP_LOGI(setup, "Initialized BME680 Sensors");
}

int samplingRateBME680TemperatureTask = 3000;

void bme680_temperature_task(void *pvParameters)
{
  // as long as sensor configuration isn't changed, duration is constant
  uint32_t duration;
  ESP_ERROR_CHECK(bme680_get_measurement_duration(&env_sensor, &duration));

  TickType_t last_wakeup = xTaskGetTickCount();

  esp_err_t res;
  while (1)
  {
    time(&now);
    // trigger the sensor to start one TPHG measurement cycle
    if ((res = bme680_force_measurement(&env_sensor)) == ESP_OK)
    {
      // passive waiting until measurement results are available
      vTaskDelay(duration);

      // get the results and do something with them
      if (bme680_get_results_float(&env_sensor, &valuesTemp) == ESP_OK)
      {
        ESP_LOGI(bme, "BME680 Temperature: %.2f °C\n", valuesTemp.temperature);
      }
    }
    // passive waiting until 1 second is over
    vTaskDelayUntil(&last_wakeup, samplingRateBME680TemperatureTask / portTICK_PERIOD_MS);
  }
}

UncleRus, 2021-06-16:
The driver may be calculating the `duration` value incorrectly. Try adding one or more ticks to it.
```C
...
ESP_ERROR_CHECK(bme680_get_measurement_duration(&env_sensor, &duration));
duration += 10;
...
```
By the way, I ask you to use the appropriate markdown tags to frame the source code in the comments - this will greatly improve its readability.

ameerhazo, 2021-06-16:
Hi @UncleRus, thanks for the suggestion. I'm sorry about the code. I'll do better next time thanks for your help.

ameerhazo, 2021-06-17:
Hi @UncleRus, the problem still persists even after I added 100 ticks to the duration of the delay. Could it be that that my hardware is the problem? The thing is that when I started using this library a 2 weeks ago, I'm still getting readings from the sensor. I've even asked created an issue here asking why I'm not getting any pressure reading which through your help got fixed. 

Is there a way maybe to reset the sensor. I'll try and check the sensor with Adafruits's library to see if its really working.

UncleRus, 2021-06-17:
> Could it be that that my hardware is the problem?
Yes, it could. But I am more inclined to think that the problem not with the device, but with some incomprehensible set of parameters. Resetting the BME can help, and it will be a good indicator.
Well, for now, I'm going to study the datasheet, maybe there is something on this topic.

UncleRus, 2021-06-17:
And how about other libraries? Does sensor work with them?

UncleRus, 2021-06-17:
@ameerhazo What's the value of `samplingRateBME680TemperatureTask` in your code? Is it more than 3 seconds?

UncleRus, 2021-06-17:
Try adding this line to your code right before triggering the measure:
```C
...
dev->meas_started = false;
// trigger the sensor to start one TPHG measurement cycle
if ((res = bme680_force_measurement(&env_sensor)) == ESP_OK)
....
```

UncleRus, 2021-06-17:
PS Remote debugging is so much fun :)

ameerhazo, 2021-06-17:
@UncleRus yup, makes sense ill try that out. Hehe sorry for taking your time. My code is the exact copy of the example code. So if it works for you and your BME680 and not mine, its probably a hardware problem on my side. Thanks again for the help.

UncleRus, 2021-06-17:
> Hehe sorry for taking your time.

Please, no need to apologize - this is my project and I am happy to do it. 
The bottom line is that I do not have a bme680 - I accidentally burned mine by feeding 12V to the SDA output :)

ameerhazo, 2021-06-17:
@UncleRus oh, RIP to that small boi :'-(

ameerhazo, 2021-06-17:
@UncleRus so, `env_sensor.meas_started = false;` fixed it. I'm getting readings now. Unfortunately, the readings are for some reason constant at 32.43 degrees celsius. Its a pretty hot day today in Dortmund Germany. I've tried to blow some cold air on the sensor but the temperature isnt going down :(

UncleRus, 2021-06-17:
> Unfortunately, the readings are for some reason constant at 32.43 degrees celsius. 

Yep, this is a problem I read about on the Bosch forum, but with the original Bosch driver. One of the forum members somehow fixed this by setting I2C clock to an incredibly low value, something around 1 kHz ... Quite odd.

You need to try with Ardino lib. Let's see if anything changes.

ameerhazo, 2021-06-17:
@UncleRus yea I'll try the Adafruit library and I'## tell you how it goes.

UncleRus, 2021-06-23:
@ameerhazo Any news?

ameerhazo, 2021-06-24:
Hi @UncleRus, yeah I've just tested it on with the Adafruit BME680 library and got the same constant readings. Its kaputt! The sad thing is I just spent 20 euros for a new BlueDotSpace BME680 thats dead on arrival and suffers from the same problem.

ameerhazo, 2021-06-24:
Also I'm using the little bro BME280 and having the same problem with this library (same constant value). I tested it with Adafruit's library and its functional. Weird.

ameerhazo, 2021-06-26:
@UncleRus nothing wrong with the BME280 driver, it was my dodgy wiring all along. Btw do you have any experience with the BME680 BSEC library from Bosch? I saw that their custom APIs use the I2C write and read as a callback for their functions. I wonder if we can still use the thread safe i2c_dev read and write function there as a callback function.

closed 2021-06-26 by ameerhazo