UncleRus/esp-idf-lib #219: Card reader reversed

silardg, 2021-06-16:
**Device type**:

- [x] ESP32
- [] ESP8226

**Framework version**:
- [x] Platformio 

**Describe the bug**:
Currently the wiegand example fills the buffer in this way:
- everything that is captured from wiegand is written into the buffer (the parity as well)
- it starts from the first 8bits and the first bit, hence the: `reader->buf[reader->bits / 8] |= value >> (reader->bits % 8);`

I'm using an android phone to read the cards, which is totally different from the one that the wiegand example is reading.
I can confirm this, because I traced the wiegand protocol on an oscilloscope, and the android phone is following the wiegand as it should. 

Long story short, I had to change the example code to start from the 4th 8bits and from the 7th bit and also ignore the first bit and put a 0x1 to the end, resulting in a good read. 

```
value = d0 ? 0x01 : 0;

if(reader->bits != 0) {
    reader->buf[3 - ((reader->bits - 1) / 8)] |= value << ((8*(((reader->bits - 1) / 8) + 1)) - reader->bits); 
}

if(reader->bits == 31) {
    reader->buf[0] |= 0b00000001;
}

reader->bits++;
```

I have hard coded the 3 (4x8bits = 32bit data). What the above does is what I described above.

I can do a pull request or give a better explanation.

UncleRus, 2021-06-16:
`wiegand` component doesn't decode protocols. Its purpose is to read and output raw data entirety, including parity bits, etc. It is up to the application using this component to decode this raw data. The reason for this is pretty simple: now there are more than a hundred protocols that are used on top of wiegand.

silardg, 2021-06-16:
Makes sense. Keep up the good work, either way, if someone is searching for an example, they can find it in this issue.

closed 2021-06-16 by silardg